marcone/teslausb #167: [tip] If you have issues with car not mounting TeslaUSB

millercentral, 2019-08-16:
I had an issue where the car would frequently not mount the TeslaUSB after leaving my house (note, mine does not have battery backup, and gets power from the USB sleeve attached to the car data port). In trying to figure out what was going on, I found that the RasPiZ was running fine (could telnet to it, no significant error messages), sitting in its "waiting for archive to become available loop". 

I took a somewhat brute force approach to the problem, and added a check_if_usb_gadget_is_mounted function call to the end of the wait_for_archive_to_be_reachable loop. Now with this change, I've had 100% success with the device mounting after the car wakes up. 

I have no idea if this is the right thing to add to the main codebase, but this hack seems to have fixed my only issue with the device. Just FYI in case anyone else faces something similar.

rggnkmp, 2019-08-16:
How & where did you add this?

millercentral, 2019-08-16:
1. While the device is connected to Wifi, connect to it with a SSH client (I use Putty on Windows) and log in (default username: **pi**, default password: **raspberry**)
2. Type **sudo -i** to act with superuser privileges. 
3. Type **cd /root/bin/** to change to the directory where the core teslausb logic is stored.
4. Type **remountfs_rw** to remount the filesystem as editable. Warning: any changes you make from here will have a permanent effect.
5. Type **nano archiveloop** to open the main program script in a text editor.
6. Scroll down until you find a function called "wait_for_archive_to_be_reachable()". Within that function scroll down until you see a line that says "sleep 1"
7. Just before that line, press return and add a new line with the same indentation and type **check_if_usb_gadget_is mounted** on that new line
8. Type CTRL+O to save the change, then CTRL+X to exit.
9. On the command prompt, type **reboot** to restart the teslausb. This will end your ssh session.

Note, I called this a brute force method because what this will do is force the teslausb to check once a second while there is no wifi/archive to see if the device is connected to the car, and if not, try and make it connect if it is not. If it _is_ connected, then the check is very lightweight, it simply looks for the presence of a devicename in the filesystem. But if there is some problem connecting it will keep retrying once a second forever. So far, this hasn't been an issue for me.

One other note to remember that tripped me up at first: While the teslausb is in the process of transferring content to your archive (ie, while connected to Wifi with access to the archive share), it will disconnect the usb device from the car. That is normal an intended. I typically see the USB device reconnect about 5 seconds after l drive away from my house and the car is out of range of my wifi.

I don't know why I need this hack and other people don't, I'm just happy to have a solution that works for me. I'm now working on using the IFTTT transfer complete notification to trigger a tesla_dashcam video processing script job on my server to combine the 3 camera views automatically.

marcone, 2019-08-17:
I don't know why the gadget drive would get disconnected like this for some people.
@kdvlr had a similar problem and contributed a workaround (https://github.com/marcone/teslausb/issues/51, https://github.com/marcone/teslausb/pull/74), but that workaround was relatively harmless, because it would reconnect the drive just once after leaving wifi range.
With your change, I'd be concerned about potentially multiple disconnects and reconnects occuring while the car is actively trying to write to it, so you should probably look into why it disconnects in the first place, rather than trying to work around the issue.

closed 2019-08-17 by marcone

millercentral, 2019-08-18:
Yes, my challenge is similar to 74. I don't have an issue with disconnect, but rather with it not mounting consistently after the car wakes up from sleep. The problem I had was if it hasn't mounted by the time archiveloop gets to wait_for_archive_to_be_reachable, it won't mount at all because the script sits in that loop until wifi and access to the archive share return. 

I've never had a problem with the drive disconnecting while driving - the only time it disconnects is when the archiving process starts, or when the car goes to sleep kills the power to the raspi.