marcone/teslausb #171: cloud storage

chiwalfrm, 2019-08-19:
This isn't really an issue but I was wondering about the possibility of uploading the TeslaCam clips to the cloud storage for example via FTP or some other method?

The reason is I have AT&T ConnectCar hotspot which only costs $20 per month for unlimited data.  It would also mean I don't have to wait to get home to get the clips archived to my Windows shared drive.

marcone, 2019-08-19:
The cloud storage part of your question is covered by rclone (see https://github.com/marcone/teslausb/blob/main-dev/doc/SetupRClone.md), however it sounds like you're really asking about continuous backups, and that's not something that's currently supported.

k1asa, 2025-02-14:
Now teslausb uses the presence of a Wi-Fi connection as a trigger to start backing up the event video.
And while teslausb is backing up the event videos, the connection from the car to the file system must be stopped to prevent file errors.
This prevents event video saving.

So it is difficult to think of an alternative to this trigger in the diverse uses that exist when using a hotspot to have a constant connection.

Is this what this issue is about?

----
Translated with DeepL.com (free version)