marcone/teslausb #172: rclone can't write config file

orzelig, 2019-08-20:
trying to use rclone following the instructions here and rclone fail to write the file. Folder permissions looks ok - any suggestions?

marcone, 2019-08-20:
Not enough information.

closed 2019-08-20 by marcone

orzelig, 2019-08-20:
Failed to save config after 10 tries: Failed to create temp file for new config: open /root/.rclone.conf937207973: read-only file system

orzelig, 2019-08-20:
root@teslausb:/# ls -l
total 69
drwxr-xr-x   2 root root    26 Aug 17 05:28 backingfiles
drwxr-xr-x   2 root root  4096 Aug 17 05:29 bin
drwxr-xr-x   3 root root  3584 Jan  1  1970 boot
drwxr-xr-x  15 root root  3540 Nov  3  2016 dev
drwxr-xr-x  85 root root  4096 Aug 17 05:35 etc
drwxr-xr-x   3 root root  4096 Jun 15 18:24 home
drwxr-xr-x  16 root root  4096 Jun 15 18:44 lib
drwx------   2 root root 16384 Jun 15 22:26 lost+found
drwxr-xr-x   2 root root  4096 Jun 15 17:59 media
drwxr-xr-x   3 root root  4096 Aug 17 05:28 mnt
drwxr-xr-x   5 root root  1024 Aug 20 05:10 mutable
drwxr-xr-x   3 root root  4096 Jun 15 18:24 opt
dr-xr-xr-x 106 root root     0 Jan  1  1970 proc
drwx------   5 root root  4096 Aug 17 05:35 root
drwxr-xr-x  24 root root   760 Aug 20 04:42 run
drwxr-xr-x   2 root root  4096 Aug 17 05:29 sbin
drwxr-xr-x   2 root root  4096 Jun 15 17:59 srv
dr-xr-xr-x  12 root root     0 Jan  1  1970 sys
drwxrwxrwt   8 root root   200 Aug 20 05:17 tmp
drwxr-xr-x  10 root root  4096 Jun 15 17:59 usr
drwxr-xr-x  10 root root  4096 Aug 17 05:30 var

orzelig, 2019-08-20:
root@teslausb:~# ls -l
total 16
drwxr-xr-x 2 root root 4096 Aug 17 05:28 bin
drwxr-xr-x 2 root root 4096 Aug 17 05:13 rpi-sync
-rw-r--r-- 1 root root   43 Aug 17 05:28 teslausb.conf
-rwxr-xr-x 1 root root  138 Jun 15 22:27 teslausb_setup_variables.conf

orzelig, 2019-08-20:
sorry for pasting all the info but it's really really weird..

marcone, 2019-08-20:
> Failed to save config after 10 tries: Failed to create temp file for new config: open /root/.rclone.conf937207973: read-only file system

Well there's your problem. You need to make /root writeable. 
But before that you may want to install a current version of rclone, because it looks like you're using an old one (perhaps you did "apt install rclone" instead of installing the current version from rclone.org?)

orzelig, 2019-08-20:
root@teslausb:/# chmod 770 root
chmod: changing permissions of 'root': Read-only file system

Can't install update because it's read-only file system

orzelig, 2019-08-20:
fixed using /root/bin/remountfs_rw

Checking if rclone will work now.