marcone/teslausb #173: Not able to see CAM/Music folder in windows

vectorvirge, 2019-08-22:
So I've reflashed and have redone the one-step-setup numerous times.  Read the related issues on the GitHub but still can't figure out why Windows can't see the Cam/Music drives.  I did the following:
sudo -i
/root/bin/setup-teslausb selfupdate
/root/bin/setup-teslausb diagnose > /mutable/diagnostics.txt
and attached the diagnostics file.  

I made sure i'm connecting the Pi Zero W w/ the data port as well.  

[diagnosis.txt](https://github.com/marcone/teslausb/files/3530633/diagnosis.txt)

marcone, 2019-08-22:
Are you sure your USB cable is good (and not a charge-only cable), and that your PC's USB port is good? Does Device Manager in Windows show the Pi?
The diagnostics show the cam and music drives published, but dmesg doesn't show that anything's connected to them. The usual cause of this is a bad or charge-only USB cable.

vectorvirge, 2019-08-22:
So tried about 5 different cables and found one that wasn't a charge only cable - that solved it.  Thanks!

closed 2019-08-22 by vectorvirge