marcone/teslausb #174: Stuck at 2 blinks even though everything looks ok?

baggachipz, 2019-08-23:
Diagnostic: https://termbin.com/zjz6

According to [testlausb-headless-setup.log,](https://termbin.com/c64f) the process finished successfully. After all, it mounts as "CAM" on usb with a "TeslaCam" folder. I can ssh into tit just fine as well. 

However, the LED is still doing two blinks which suggests that perhaps not everything finished successfully. I ran a re-install because I noticed that the archive hadn't been copied for the last couple weeks and so just wanted to wipe and start over for simplicity's sake. Is there something wrong/missing here? Any help is greatly appreciated.

DMBlakeley, 2019-08-23:
Take a look at the Wiki link. Double blink indicates that pi is operating and connected to WiFi.

marcone, 2019-08-23:
Maybe the confusion is that the setup script uses 1, 2, 3, 4 and 5 blinks to indicate completion of certain stages of setup.
What you're seeing now is just the Pi indicating it is running normally and connected to wifi. If you go for a drive you should see it change to 1 blink. When you come back home it will blink rapidly for a little while, then settle in the 2-blink pattern again.

closed 2019-08-23 by marcone

baggachipz, 2019-08-23:
Great, thank you! Really appreciate you maintaining this repo.