marcone/teslausb #175: USBstorage towards tesla comes and goes

joverdijk, 2019-08-23:
Hi

I am trying to get some webcam stills out of the RecentClips dir, by mounting read-only the cam-disk.bin and unmounting it after I got last few frames from latest clip with ffmpeg.

fstab line to mount:
/backingfiles/cam_disk.bin /mnt/rocam vfat ro,utf8,noauto,users,umask=000,offset=1048576 0 0
I run a cronjob grabbing the last few frames from all 3 cams to jpg and scp them to a remote server (teslausb rpi is connected to 4G via wifi tethering).

basically this works, but I noticed since I'm doing this, the Camera icon and also my music USB disconnects and comes back to the car. It's quite random, not only on the 10 minute interval that I do the cronjob with the readonly-mounting and unmounting of the cam.bin 
It's quite random, sometimes it works for 30-40 minutes, then it resets 3 times in 10 minutes. It seems worse when listening to mp3 on the music.disk

Any idea what is causing this? 
Or a better way to access the camera-files from the raspberry, while the same image stays as storage available to the car at all times?

I was watching the scripts and tried making a snapshot of some sort, but this does not work:

root@tesla-pi:/backingfiles# cp --reflink=always /backingfiles/cam_disk.bin /mnt/camsnap.bin
cp: failed to clone '/mnt/camsnap.bin' from '/backingfiles/cam_disk.bin': Invalid cross-device link
root@tesla-pi:/backingfiles# cp --reflink=always /backingfiles/cam_disk.bin /mnt/rocam/
cp: failed to clone '/mnt/rocam/cam_disk.bin' from '/backingfiles/cam_disk.bin': Invalid cross-device link

Some help would be nice! Thanks!

marcone, 2019-08-23:
A reflink/snapshot needs to be on the same filesystem, so you need to make your copy in /backingfiles (this is why it said "Invalid cross-device link")

As for why what you're doing causes the drive to disappear from the car, check dmesg and see if maybe you're overloading the Pi too much and creating under-voltage conditions.

joverdijk, 2019-08-23:
I already considered the under-voltage thing, so I took a 12V plug 2.1Amp's and a microusb cable and plugged it into the power-port of Rpi Zero. So, power from 12V @ 2 amp and second port of Zero connected to car, but also in that condition problems were there.
Also, overloading you'd suspect to occur at cronjob times (every 10min), but i could not correlate it to that ;-/
Will try to dmesg shortly.. if car goes to sleep, PI is cutoff, so no logging in dmesg right now

pleasantone, 2019-10-02:
@marcone I know you've said you use a battery system, what exactly are you doing (a data-only no-charge cable from Tesla->USB and a charge-through battery going to PWR on the Pi?). What models?

marcone, 2019-10-02:
I used to use battery, but now I'm using an ODB2-to-USB power adapter. Unfortunately the car doesn't seem to follow the requirement of having an always-on power source present in the OBD2 connector, so it still loses power sometimes.

pleasantone, 2019-10-02:
@marcone Dumb questions but where is the odb2 port and which adapter did you buy? Also, again, are you using a data-only cord since the +5v points are tied together everywhere? Did you make the data-only cord or buy one? If you made it, did you just cut the red wire?

This stuff would be really great in the wiki so we are using a setup as similar to yours as possible?

marcone, 2019-10-02:
On the S the OBD2 port is in the driver footwell. It's my understanding that the 3 does not have an OBD2 port.
For the data-only cable, I put a piece of packing tape over the contact in the connector ([example](http://folgerforum.com/t/quick-and-simple-non-destructive-mod-for-a-data-only-no-power-usb-cable/2086)). I think the data-only cable is only important if you're using a power source other than the car's USB ports though. If you're using two USB ports to power the Pi then I don't think you need a data-only cable.

pleasantone, 2019-10-03:
Thanks, I switched to two USB ports and the sentry option and it’s robust now!

closed 2019-10-03 by marcone