marcone/teslausb #176: Question: What is the best way to "peek" into the CAM partition over ssh?

pirin, 2019-08-23:
I am trying to find a way to SSH into the PI and access the CAM partition to check that Tesla is properly writing files there. For some strange reason my 3 sometimes decides to start outputting only 500 byte files and needs a hard reset to start working properly again. I can't see that this is happening unless I connect it to a computer and see the CAM contents.
 
I know the CAM partition is not accessible while connected to the car, so I was hoping to find a way to SSH, mount-check-unmount without "breaking" the archive loop. If somebody can point me to the proper set of commands (or where in the archive loop that happens), I will highly appreciate it.

joverdijk, 2019-08-23:
Its a known bug when you leave sentry mode on for more than like 24 hours, the car begins to write 595 byte files. Resetting does not help, only thing that helps is turning sentry off and let the car go to sleep. After that, problem is gone.

You can probably mount the cam partition just by mounting it 'mount /mnt/cam' that uses the entry in /etc/fstab.

I dont know if this breaks anything. I myself am also looking into a good sollution to have readonly access to the CAM-partition while the cam-partition is actually being written files to by the car.
So i'm also seeking for reliable way to 'peek' into that partition.
Anybody knows good sollution for this?

marcone, 2019-08-24:
> So i'm also seeking for reliable way to 'peek' into that partition.
> Anybody knows good sollution for this?

Use snapshots. Have a look at the smb_on.sh, mount_snapshot.sh and smb_off.sh scripts.