marcone/teslausb #193: Tesla unable to write to CAM disk

tgerighty, 2019-08-28:
Using the latest pre-built image (teslausb-20190615) configured on 400Gb San Disk Ultra as a single disk (100% CAM). Connected to Wifi and Archiving fine (tested by copying Saved Clips from old USB stick to CAM). CAM disk does appear on my Mac.

The issue is that the Tesla (MX) never writes to the CAM disk - no Sentry recordings and I have the Dashcam "X".

Any ideas?

diagnostic.txt
https://pastebin.com/qD4kNr0p

tgerighty, 2019-08-28:
> Not sure why you would get the 'X', but at the time you took the diagnostics it was archiving recordings, albeit very slowly:
> 
> Wed 28 Aug 14:22:47 BST 2019: Moving '2019-08-27_18-12-05/2019-08-27_18-04-08-front.mp4'
> Wed 28 Aug 14:25:15 BST 2019: Moved '2019-08-27_18-12-05/2019-08-27_18-04-08-front.mp4'
> 
> The fact that CAM shows up on the Mac but not in the car would seem to indicate a problem with the car. Have you tried resetting the MCU?
> 
> One tip in case you end up reinstalling teslausb: for a large card like that, don't use 100% CAM. It's just not useful. Much better to use something like 16GB so teslausb has lots of free space to cache recordings that are older than an hour.

Thanks - the archiving is working because I copied files from my old usb stick to it to check/test that archiving was working. All functions appear to work. except the car not recognising it. Will try a MCU reboot.

Thank you!

tgerighty, 2019-08-28:
> I would check the format of the drive on the Mac. Needs to be FAT32 to work on Tesla, exFAT will not work. As @marcone suggested I would set the camsize to a fixed value.

Thanks - I flashed it using the pre-built image and let it create the CAM partition so presume it defaults to FAT32?

tgerighty, 2019-08-28:
> > > This is something I've been working on for a while that I just merged to the main branch a few days ago. Teslausb now takes hourly snapshots of the CAM disk image, which allows keeping "RecentClips" recordings older than an hour. There needs to be enough free space to keep those snapshots though.
> > 
> > 
> > Ah-haha. Hmmm. Does this mean _not_ creating two partitions as I did, and instead giving CAM, say, 55% and allow the rest for MUSIC? I'm realizing that, perhaps, 128 Gb may not be big enough, and a 256 Gb card may be a more appropriate size, given your new strategery.
> 
> If you allow the rest for MUSIC, then there still won't be any/much free space. You'll need to specify CAM and MUSIC sizes such that they don't add up to 100%. I don't need several days worth of music in my car, so I have music set to 2G, and cam to 16G (you can specify absolute sizes, it doesn't have to be a percentage). On a 128GB card that leaves enough free space to keep over 17 hours worth of "RecentClips" (that's 17 hours worth of actual recordings, not just the most recent 17 hours)
> 
> (also, nitpick: MUSIC and CAM are not actually partitions. They're two separate disk images, stored on a single partition on the Pi, which the attached computer/car will see as two separate drives, not as two partitions on the same drive)

Ah so if I don't want music, I set CAM to 16G; Music to 0% then the remainder is all available for recordings?

marcone, 2019-08-28:
I don't really want to start adding random packages that have no direct bearing on teslausb. Those can be installed after the fact by those that need them.

sTevefor3, 2019-08-29:
> I don't really want to start adding random packages that have no direct bearing on teslausb. Those can be installed after the fact by those that need them.

ok.

tgerighty, 2019-08-29:
> > Ah so if I don't want music, I set CAM to 16G; Music to 0% then the remainder is all available for recordings?
> 
> Correct.
> 
> > Can they be resized from the running device or do I need to re-flash?
> 
> You don't need to reflash, but it's probably the easiest.
> Without reflashing, you'll need to kill archiveloop, unload the g_mass_storage module, unmount all disk images and snapshots, delete them, then re-run setup. I should probably make a setup option that does all that.

Reflashed to 16G CAM, 0% Music. Still the same - Mac sees the CAM drive; Car doesn't (No sentry or Dashcam recordings), TeslaCam folder is empty. Car MCU rebooted twice. Will try someone else's car next.

DMBlakeley, 2019-08-29:
Just a check, did a standard FAT32 formated USB flash drive with a TeslaCAM folder work on your Tesla?

tgerighty, 2019-08-29:
> Just a check, did a standard FAT32 formated USB flash drive with a TeslaCAM folder work on your Tesla? Just want to make sure it is not an issue with your Tesla rather than the Pi.

Yes - standard FAT32 disk works fine.

bahree, 2019-08-29:
So I have been having some issue since the latest update and before that things have been working OK. I got a 128GB card and everything set to CAM, with nothing to Music. And last 2 days or so, have to reboot the MCU multiple times, or the pi, or both to get it to recognize in the car (as a drive to save the recording).

Is this anyway related to the change [the comment here](https://github.com/marcone/teslausb/issues/193#issuecomment-525796203) and this now being too large? Sorry if I am misinterpreting it.

DMBlakeley, 2019-08-29:
Did you connect the Pi to your Tesla with the same cable that you used to connect to your Mac? Is the Pi actually booting with either 1 flash if not connected to wifi or 2 flashes if connected to wifi? (Sorry if questions already asked.)

tgerighty, 2019-08-29:
> Did you connect the Pi to your Tesla with the same cable that you used to connect to your Mac? Is the Pi actually booting with either 1 flash if not connected to wifi or 2 flashes if connected to wifi? (Sorry if questions already asked.)

Same cable and tried a different one. Booted fine to two flashes and I can ssh to it from the house (so on wifi and logs show its mounted the drive/waiting for archive etc).

tgerighty, 2019-08-29:
> > Did you connect the Pi to your Tesla with the same cable that you used to connect to your Mac? Is the Pi actually booting with either 1 flash if not connected to wifi or 2 flashes if connected to wifi? (Sorry if questions already asked.)
> 
> Same cable and tried a different one. Booted fine to two flashes and I can ssh to it from the house (so on wifi and logs show its mounted the drive/waiting for archive etc). Just never appears to the car and doesn't get anything written to it.

DMBlakeley, 2019-08-29:
Curious what version of software is the Tesla at? I am on 2019.28.3.1 and all seems to be working correctly.

tgerighty, 2019-08-29:
> Curious what version of software is the Tesla at? I am on 2019.28.3.1 and all seems to be working correctly.

I'm hoping its that 2019.20.xx but had an update arrive tonight. Lets see in the morning!

tgerighty, 2019-08-30:
> > Curious what version of software is the Tesla at? I am on 2019.28.3.1 and all seems to be working correctly.
> 
> I am on 2019.20.xx but had an update arrive tonight. Lets see in the morning!

Sadly same result - 2019.28.3. Car never sees the disk so never writes to it. Pi sits with single or two flashes (depending on if its close to home/on wifi). I can ssh and see the logs all show its running ok.

- Does it matter which port you plug into (MX)? I tried both.
- I've tried two different data/charge cables connected to the USB on the Pi.
- I've even tried booting the Pi connected to 2.4A usb via the 12v (connected to the Pi's regular power port) and then connecting a data/charge cable connected to the car via the Pi USB port.
- Regular USB stick works...

Any ideas of things I can try?

DMBlakeley, 2019-08-30:
On my M3 I have always plugged the Pi or USB flash drive into the front left port. I believe that either front port can be used but the USB ports for the back seat are for charging only.

As a regular USB works, the Pi seems to boot normally, you have tried multiple cables, and you have tried providing power to the Pi through the power port and provided a separate USB data connection the only thing that I can think of is that you have a defective Pi. This doesn’t totally make sense as the Pi CAM folder connects to your Mac.

Will be curious if you do figure it out.

tgerighty, 2019-09-01:
So its certainly the car. Just tried a new Pi with another Micro SD Card (Sandisk Ultra Endurance 128Gb) and same result. Will be trying it Tuesday in a friend's X.

tgerighty, 2019-09-02:
One difference is that I have a MX MCU1/AP2.5 so it may be that combo has issues. Testing another MX tomorrow.

awisur, 2019-09-04:
I have a MS MCU1/AP2.5 with same result.
It mounts on PC, but does not show icon on car.
I also tested read Write speed on PC, with USB 3.0 cable i only get 2MB per second write speed. Get around 20 when reading from the CAM disk. SD Card is Samsung EVO plus 128, read:100/write:90, and I get close to those speeds when reading/writing with a SD card reader.

awisur, 2019-09-04:
Herer is my diagnostics BTW: https://pastebin.com/9eGw5JVS

tgerighty, 2019-09-05:
Ok. So I have a workaround.

Got the idea from this:
https://raspberrypi.stackexchange.com/questions/68110/pi-zero-g-mass-storage-not-recognized-by-some-computers-until-after-another-comp
and this:
https://www.spinics.net/lists/linux-usb/msg157730.html

It appears that *some* devices fail to see the g_mass_storage as a USB disk at all. At this stage I am presuming that the MCU1 Nvidia CPU is one of those devices.

Workaround:

- Power the Pi-zero sign the power port from either the car or a USB battery'
- Connect the Pi-zero USB port to a PC or Mac (I have only tried on a Mac but presume it works)'
- Boot up. Wait for CAM/MUSIC drives to appear.
- Disconnect Pi-zero USB port (keep Pi powered and running) and reconnect to car.

Car has now seen the disk, recorded its first Sentry video and archived it to my NAS.

Allram, 2019-09-05:
Unfortunately this will only work until the mcu reboots/turn off the USB power. When it boots up again we're into the same problem (unless the USB hub has a lot of power and can drift the pi for a long time).

awisur, 2019-09-05:
I have tried a lot of stuff this evening, and I have a great workaround :D
Connect the RPIz through a USB hub. This way it will work with only one cable, and it will show up with CAM icon, no need for any battery or dual power either.

It just works!

tgerighty, 2019-09-06:
Awesome. A simple plain hub or a powered one?

awisur, 2019-09-06:
I use an unpowered hub with one usb cable from the tesla, and one usb cable from the hub to the RPIz. Plain and simple :)

tgerighty, 2019-09-06:
Can you share what hub you are using? I don’t get it working with a startech one.

awisur, 2019-09-06:
I use this as an unpowered USB hub.
https://www.ebay.com/itm/Dock-Adapter-7-Port-USB-2-0-HUB-Converter-with-AC-Power-For-PC-Laptop-Computer/273948837851?_mwBanner=1&ul_ref=https%3A%2F%2Frover.ebay.com%2Frover%2F0%2F0%2F0%3Fmpre%3Dhttps%253A%252F%252Fwww.ebay.com%252Fulk%252Fitm%252F273948837851%26rvr_id%3D0%26rvr_ts%3D0711b95716d0a4e908b5d9cdffc5ba23&ul_noapp=true&pageci=aa1927b9-911f-4329-b188-689dfbeac40d

tuupos, 2019-09-07:
Thanks Awisur! I will order the same HUB. Unfortunately i cannot get it to work with a HUB lying around..

This one:
https://www.bol.com/nl/p/maxxter-usb-hub-3-poorten-snelle-data-overdrachtsnelheid-tot-480-mbps-zwart/9200000100062622/?Referrer=ADVNLGOO002013-G-49170246675-S-469889519734-9200000100062622&gclid=EAIaIQobChMI1eGwwem-5AIV0ud3Ch3c6Qn_EAQYASABEgJ9zPD_BwE

Why does one HUb work and the other one dont??

mattaran, 2019-09-13:
I had the same issue in my early 2018 Model S (MCU1/AP2.5) and was able to get it working by using an "Inateck HB4012" Ultra Slim 4-Port USB 3.0 hub I had laying around.  This is a un-powered hub. The connection is as @tuupos listed above.

JohnPhilipBiggs, 2019-09-26:
I am now having the same problem with my late 2017 Model S (MCU1/AP2.5) after a recent upgrade to 2019.32.2.2 (da05838).  It was working fine before but now the car won't recognise the CAM disk although it still mounts fine on my Mac.  I have tried upgrading to teslausb-buster-20190922 - (64GB Samsung PRO Endurance, 16GB CAM 0GB MUSIC) - boots just fine, I get the 2 flashes and can ssh in to it from WiFi -  but no DashCam icon :-(

I am wondering if a USB Hub (Anker 4 port UltraSlim?) will solve my problem - I have some doubt as I didnt need one before - so why would a SW upgrade mean I need one now?

tuupos, 2019-10-09:
I ordered the USB hub as awisur mentioned above and it works perfectly in my model S MCU1/AP 2.5. Red recording button appears!!
The only thing is my car complains after a few minutes that the drive speed it's to low for recording. But that's another issue I will dive into

tuupos, 2019-10-13:
@awisur do you still have a functional dashcam after the V10 update? Mine will only work for 1,5 minutes before going into a grey X icon. I read a lot about the V10 update killed for most people the function like me but other report that it still works. I just want to know if it is a tesla software issue or a HUB issue not getting the speeds in needs through the HUB.
Also wondering if it works with your setup (now equal to mine with same HUB) which exact SD card do you have and what are your partitions/drive sizes?  Thanks in advance

tuupos, 2019-10-27:
@sparks3875392 is it functioning even after the v10 update? I cannot get more then 1.5 minutes of red icon before the gray x appears... tried all different cables, versions and sd cards available...

sparks3875392, 2019-10-27:
Yep working for me on v10 2019.32.12.3 MCU1/AP2.5 MS

tuupos, 2019-10-27:
@sparks3875392 I dont get it why it doesnt work in my tesla.
Can I ask how did you setup your SD card? Stretch or buster? partition sizes? vm dirty flags ? updated rpi image? cam and music sizes? Maybe if I can copy your configuration I can narrow down the issue in my tesla. Are you using a Rpi0 or Rpi 4? Thanks in advance

sparks3875392, 2019-10-27:
Pi Zero W buster from the new image, 256Gb 

camsize=24G
musicsize=50G

Didn’t understand the cam dirtyflags. It always boots dirty if the power has been off . Sometime I have music mounted but dashcam failed.  haven’t worked out why

tuupos, 2019-10-27:
I got it working!!!!!
Ast first I used the same usb hub as awisur had. He claimed to have success with it. But this hub gave me just 1.5 minutes of success.  I now used another hub similar to sparks3875392s. Works like a charm with stretch, latest image, 24G cam,  4G music, on a average 64GB sd card. No other alterations needed. 
Tried every sd card,  every dirty flag setting, every update of pi and different partitionsizes and settings and all along it was a simple 2.0 hub....

closed 2019-11-02 by marcone