marcone/teslausb #194: x509: certificate has expired or is not yet valid

DaveWitchalls, 2019-08-28:
Hi,

Finally making some progress in getting the Google Drive rclone working, for the first time it's actually tried to copy files to google drive and got the above error (x509: certificate has expired or is not yet valid).

I set the token up when doing the setup, so that should be OK? The System time is now set with rdate to Europe/London 

This was the error when I tailed the archiveloop log file:

root@teslausb:/# cd mutable
root@teslausb:/mutable# tail archiveloop.log
2016/11/03 17:17:54 ERROR : Attempt 3/3 failed with 2 errors and: couldn't list directory: Get https://www.googleapis.com/drive/v3/files?alt=json&fields=files%28id%2Cname%2Csize%2Cmd5Checksum%2Ctrashed%2CmodifiedTime%2CcreatedTime%2CmimeType%2Cparents%2CwebViewLink%29%2CnextPageToken&includeItemsFromAllDrives=true&pageSize=1000&prettyPrint=false&q=trashed%3Dfalse+and+%28%27root%27+in+parents%29+and+%28name%3D%27TeslaCam%27%29+and+mimeType%3D%27application%2Fvnd.google-apps.folder%27&supportsAllDrives=true: x509: certificate has expired or is not yet valid
2016/11/03 17:17:54 Failed to move with 2 errors: last error was: couldn't list directory: Get https://www.googleapis.com/drive/v3/files?alt=json&fields=files%28id%2Cname%2Csize%2Cmd5Checksum%2Ctrashed%2CmodifiedTime%2CcreatedTime%2CmimeType%2Cparents%2CwebViewLink%29%2CnextPageToken&includeItemsFromAllDrives=true&pageSize=1000&prettyPrint=false&q=trashed%3Dfalse+and+%28%27root%27+in+parents%29+and+%28name%3D%27TeslaCam%27%29+and+mimeType%3D%27application%2Fvnd.google-apps.folder%27&supportsAllDrives=true: x509: certificate has expired or is not yet valid
Thu  3 Nov 17:17:54 GMT 2016: Moved 0 file(s).

Bit weird the errors are dated 2016? - Any pointers on what to check next?

Dave.

DaveWitchalls, 2019-08-29:
Thanks both! 

I’d used rdate to set the time, based on someone else’s guide to get this working, what time zone should I be setting it to? 

Or should I try and rebuild again with he latest commit? I’d used the pre tilt imagine (latest one, but still from June?). 

Thanks
Dave.

marcone, 2019-08-29:
The June prebuilt is fine. It will download the latest scripts during setup.
Timezone doesn't matter, though I'd suggest setting it to the timezone you're actually in, so that timestamps in the logs line up with your local time.

closed 2019-08-29 by marcone