marcone/teslausb #197: Force Archive via SSH?

DavidGagne, 2019-08-29:
TL;DR: Is there a command to execute via an SSH connection to "force" an archive process?

I know it's not technically supported via this fork, but I'm trying to test rsync and I'm not sure why it's not working.

Inspecting _/mutable/archiveloop.log_ shows me the log messages:

> Archiving through rsync...
> No files to archive through rsync.

But I can see that there are files in the TeslaCam directory which _should_ be getting uploaded. I know that passwordless ssh is working correctly from my Raspberry Pi to my server, so I'm trying to track down where it's failing.

marcone, 2019-08-29:
`touch /tmp/archive_is_unreachable` will simulate a disconnect/reconnect from wifi, which will trigger archiving. Note that teslausb only archives files under SavedClips, not under RecentClips (in case that explains why you're seeing files under TeslaCam but nothing is getting archived)

closed 2019-08-29 by marcone

DavidGagne, 2019-08-29:
Is the archiveloop executed by the user pi or the user root?