marcone/teslausb #198: Stuck on 2 led flashes

lifeisgrand, 2019-08-29:


lifeisgrand, 2019-08-29:
[teslausb-headless-setup.log](https://github.com/marcone/teslausb/files/3553897/teslausb-headless-setup.log)

rohan3au, 2019-08-29:
Also having this problem, stuck on two flashes. Have tried many fixes with nothing working. Any advice?

marcone, 2019-08-29:
https://github.com/marcone/teslausb/issues/35

closed 2019-08-29 by marcone

lifeisgrand, 2019-08-29:
<script src="https://pastebin.com/embed_js/t7V7NXpt"></script>

rohan3au, 2019-08-29:
Have gone through all of the problems and still not working. Pi won't even reach SSH stage, and doesn't produce a log file.

marcone, 2019-08-29:
@lifeisgrand I can't access that link ("this is a private paste")

@rohan3au then you probably specified the wrong wifi credentials. Note that special characters in SSID or password need to be escaped.

lifeisgrand, 2019-08-29:
I made the link hidden instead of private - it should work now

marcone, 2019-08-29:
@lifeisgrand that looks like a perfectly normal working install. Regarding the flashes, please see the first sentence in the wiki (https://github.com/marcone/teslausb/wiki referenced from https://github.com/marcone/teslausb/issues/35)