marcone/teslausb #201: Saved Clips separate to Recent Clips?

Jonnehs, 2019-08-29:
I dunno if I've just missed some obvious configuration somewhere, but my Pi copies basically everything to the SMB share from Recent Clips, which is the entire time the camera is recording while driving as well as your sentry events etc.

The problem is finding events because some days this can be a lot of data. Should it keep "Saved Clips" separate to "Recent Clips" in some way when transferring them? 

If I've missed something somewhere, apologies, at the moment I just get all the video transferred to the root of my SMB share with no indication if the clip is from a sentry event, a manual dashcam save button press, or just the recent clips cache of footage from driving.

If it doesn't exist, I'd propose that it preserve the indication that footage is in the "Saved Clips" folder somehow, as this would make finding sentry events when you don't know the time it may have gone off a bit easier.

carleeno, 2019-08-29:
This is strange, normally it would only copy whats in the SavedClips directory.
Which version did you install (prebuilt or manual setup) and how long ago did you install it?

Jonnehs, 2019-08-29:
two days ago, the pre-built image flash

It did the same thing before then on a previous incarnation built the same way

carleeno, 2019-08-29:
Can you share your log file (showing most recent archive) and also a screenshot of the files in your archive server? So I can get an idea of the folder structure it's copying over.

Jonnehs, 2019-08-29:
[image]

Can't do the log file until I get home and can ssh to it

carleeno, 2019-08-29:
Ok looks like you have something triggering your sentry mode quite often, maybe rain, or a light turning on/off, or people walking by while parked in a public area. Those folders are only from the SavedClips folder. If you look at the time in the filenames of those folders, you'll see that they are at irregular intervals.

The recent clips folder keeps every minute for the last hour, which means you'd only have 1 hour worth of recent clips for each time teslausb runs a backup.

Each one of the folders I see in your screenshot should have up to 10 minutes of files (30 files max), but can have less if the previous sentry mode trigger was less than 10 minutes prior.

carleeno, 2019-08-29:
...oh and the quickest way to find what triggered the event is to go to the last minute or sometimes the 2nd last minute saved in each subfolder. I use VLC and playback at 4x speed (using +/- buttons to change speed)

Jonnehs, 2019-08-29:
Sorry that is a pretty bad example screenshot because those are all saved clips in the shot, but previously there are continuous timestamps of my commutes even when I do not press the dashcam button.

You're right though, the ones since I rebuilt the image the other day seem to only be sentry events. Perhaps whatever was wrong before was fixed when I rebuilt.

carleeno, 2019-08-29:
That's ok.
Looking at the most recent versions, I can't see why that would have happened; but I agree, maybe the rebuild has fixed it.
If it happens with your current build, send more screens and the log file.

marcone, 2019-08-29:
teslausb has never backed up RecentClips.

closed 2019-08-29 by marcone

LostSoulfly, 2019-09-11:
> teslausb has never backed up RecentClips.

Might want to add this somewhereon the main readme. I was looking for a way to modify this so it would only backup my sentry mode footage instead of **all** the dashcam footage. Turns out I didn't need to worry about that :)