marcone/teslausb #202: Can SSH in via WIFI, but not USB

skyred, 2019-08-29:
Everything works as it advertises so far, if I plug in my computer, I can see the CAM drive. I can also ssh in via WIFI. But, I want to disable wifi to work on some improvement https://github.com/marcone/teslausb/issues/86 , I simply disabled my wifi radio, but still connect with USD (I can still see the CAM drive with its content), but I can't `ssh pi@teslausb.local` any more.

error: 

> ssh: Could not resolve hostname teslausb.local: Name or service not known

Then, I rebooted the Pi, I can still see the CAM drive, but still no SSH. What could cause this?

marcone, 2019-08-29:
I'm not sure I understand your question. You said you disabled wifi, but then wonder why you can't ssh into the Pi over wifi anymore?

bahree, 2019-08-29:
Maybe are you thinking of USB networking? Does that work OOB?

skyred, 2019-08-30:
I made a typo above. I meant SSH via USB networking. I thought the documentation says it is supported, no?

marcone, 2019-08-30:
As far as I know USB networking requires the g_ether module, which won't be loaded anymore once the g_mass_storage module is loaded to provide the USB drive emulation.
You would have to use the g_multi module instead if you want both mass storage and usb networking at the same time.

closed 2019-08-30 by marcone

skyred, 2019-08-30:
Thanks for the explanation, that would make sense. But, do you think using g_multi could be a feature request that you would accept a PR, if someone writes it?

marcone, 2019-08-30:
I'll certainly consider a PR. Note that this might require changing how teslausb publishes and unpublishes the mass storage device, since it currently does that by load and unloading the g_mass_storage module. If you simply replaced that with the g_multi module, you'd lose usb networking when archiving starts.