marcone/teslausb #204: Tesla unable to write/read Pi during extended sentry mode

bensechrist, 2019-08-29:
I have gotten teslausb to work with my setup before but recently after an update when leaving my car on sentry mode for more than a few hours causes an error and no events are recorded. When I return to my car an error is displayed that the Tesla cannot read the usb drive. To reenable the drive I have to unplug the Pi for a bit and plug it back in. Manually saving a clip works and is synced via rsync to my server.

bensechrist, 2019-08-29:
Diagnostics file: https://gist.github.com/bensechrist/f8f436839e248b5339f18ba7dc2f11dd

sTevefor3, 2019-09-04:
IDK if this is a solution:

I was on holiday last weekend.  The "[Stats](https://www.maadotaa.com/)" app has a "Smart Sentry Mode" feature (see pic).  

[image]

I had enabled it, and I noticed that Sentry Mode was being enabled (the app announces this on my iPhone), even though I had created a Favorite from my location (my brother's rural house) in both the car's onboard screen and in the Stats app, itself.  I disabled "Smart Sentry Mode" in the Stats app on the third day of my holiday stay. 

The Pi would "go to sleep", as it were, at random times, mostly after I had parked for a few hours.  This "sleep" mode would cause the car to dismount both CAM and MUSIC drives.  The only solution that worked was to reboot the Pi or, if I were away from my home network, unplug/replug its power.

After I disabled "Smart Sentry Mode" in the Stats app, the problem disappeared.  Thus, on my return trip, the Pi performed brilliantly and never went to sleep.  Pulled into my driveway 1230 this morning, and the Pi dutifully connected to my home WiFi, and updated video and audio.  My server shows times of transfer as late as 240AM (I've not checked the Pi's logs, yet).

Took me a few hours to debrief myself before I discovered this.  Hope this helps.

bensechrist, 2019-09-04:
I don't use the Stats app. Are you able to use Sentry mode with the latest TeslaUsb code though?

sTevefor3, 2019-09-04:
> I don't use the Stats app. Are you able to use Sentry mode with the latest TeslaUsb code though?

Yes.  My Model 3 is on 2019.32.

bensechrist, 2019-09-04:
I'm on 2019.28 right now

marcone, 2019-09-04:
@bensechrist what does `df -h /backingfiles` say?

bensechrist, 2019-09-05:
```
pi@teslausb:~ $ df -h /backingfiles 
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p3   29G   28G 1023M  97% /backingfiles
```

marcone, 2019-09-05:
I would recommend reflashing, as I think there may be a problem when upgrading a previous install (a fresh install will keep more free space in reserve).
I also recommend using a larger card.

bensechrist, 2019-09-06:
I did a fresh install and that seems to have cleared up the issue but now saved clips from the pi are not being transferred to my server. I am using rsync/sftp

```
Fri  6 Sep 14:19:06 EDT 2019: Archiving...
Fri  6 Sep 14:19:06 EDT 2019: Disconnecting usb from host...
Fri  6 Sep 14:19:06 EDT 2019: Disconnected usb from host.
Fri  6 Sep 14:19:06 EDT 2019: Checking saved folder count...
Fri  6 Sep 14:19:06 EDT 2019: Ensuring cam file is mounted...
Fri  6 Sep 14:19:06 EDT 2019: Mounting /mnt/cam...
Fri  6 Sep 14:19:06 EDT 2019: Mounted /mnt/cam.
Fri  6 Sep 14:19:06 EDT 2019: Ensured cam file is mounted.
Fri  6 Sep 14:19:06 EDT 2019: Running fsck on /mnt/cam...
Fri  6 Sep 14:19:06 EDT 2019: | fsck from util-linux 2.29.2
Fri  6 Sep 14:19:07 EDT 2019: | fsck.fat 4.1 (2017-01-24)
Fri  6 Sep 14:19:07 EDT 2019: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Fri  6 Sep 14:19:08 EDT 2019: |  Automatically removing dirty bit.
Fri  6 Sep 14:19:08 EDT 2019: | Free cluster summary wrong (543113 vs. really 553967)
Fri  6 Sep 14:19:08 EDT 2019: |   Auto-correcting.
Fri  6 Sep 14:19:08 EDT 2019: | Performing changes.
Fri  6 Sep 14:19:08 EDT 2019: | /dev/loop1: 434 files, 729687/1283654 clusters
Fri  6 Sep 14:19:08 EDT 2019: Finished fsck on /mnt/cam.
Fri  6 Sep 14:19:08 EDT 2019: There are 10 event folder(s) with 273 file(s) to move.
Fri  6 Sep 14:19:08 EDT 2019: Starting recording archiving...
Fri  6 Sep 14:19:08 EDT 2019: wake up car process is 25882
Fri  6 Sep 14:19:08 EDT 2019: Archiving through rsync...
Fri  6 Sep 14:20:39 EDT 2019: No files to archive through rsync.
```

bensechrist, 2019-09-06:
Contents of /tmp/archive-rsync-cmd.log

```
2019/09/06 14:43:19 [3399] building file list
2019/09/06 14:43:19 [3399] rsync: link_stat "/mnt/cam/TeslaCam/saved*" failed: No such file or directory (2)
2019/09/06 14:43:19 [3399] rsync: recv_generator: mkdir "/media/SideMountedHDD/TeslaUsb/2019-09-06_09-10-27" failed: Permission denied (13)
2019/09/06 14:43:19 [3399] *** Skipping any contents from this failed directory ***
2019/09/06 14:43:19 [3399] rsync: recv_generator: mkdir "/media/SideMountedHDD/TeslaUsb/2019-09-06_09-19-37" failed: Permission denied (13)
2019/09/06 14:43:19 [3399] *** Skipping any contents from this failed directory ***
2019/09/06 14:43:19 [3399] rsync: recv_generator: mkdir "/media/SideMountedHDD/TeslaUsb/2019-09-06_10-21-58" failed: Permission denied (13)
2019/09/06 14:43:19 [3399] *** Skipping any contents from this failed directory ***
2019/09/06 14:43:19 [3399] rsync: recv_generator: mkdir "/media/SideMountedHDD/TeslaUsb/2019-09-06_11-31-15" failed: Permission denied (13)
2019/09/06 14:43:19 [3399] *** Skipping any contents from this failed directory ***
2019/09/06 14:43:19 [3399] rsync: recv_generator: mkdir "/media/SideMountedHDD/TeslaUsb/2019-09-06_12-08-10" failed: Permission denied (13)
2019/09/06 14:43:19 [3399] *** Skipping any contents from this failed directory ***
2019/09/06 14:43:19 [3399] rsync: recv_generator: mkdir "/media/SideMountedHDD/TeslaUsb/2019-09-06_12-19-15" failed: Permission denied (13)
2019/09/06 14:43:19 [3399] *** Skipping any contents from this failed directory ***
2019/09/06 14:43:19 [3399] rsync: recv_generator: mkdir "/media/SideMountedHDD/TeslaUsb/2019-09-06_12-22-42" failed: Permission denied (13)
2019/09/06 14:43:19 [3399] *** Skipping any contents from this failed directory ***
2019/09/06 14:43:19 [3399] rsync: recv_generator: mkdir "/media/SideMountedHDD/TeslaUsb/2019-09-06_12-35-58" failed: Permission denied (13)
2019/09/06 14:43:19 [3399] *** Skipping any contents from this failed directory ***
2019/09/06 14:43:19 [3399] rsync: recv_generator: mkdir "/media/SideMountedHDD/TeslaUsb/2019-09-06_12-37-04" failed: Permission denied (13)
2019/09/06 14:43:19 [3399] *** Skipping any contents from this failed directory ***
2019/09/06 14:43:19 [3399] rsync: recv_generator: mkdir "/media/SideMountedHDD/TeslaUsb/2019-09-06_13-52-02" failed: Permission denied (13)
2019/09/06 14:43:19 [3399] *** Skipping any contents from this failed directory ***
2019/09/06 14:43:19 [3399] cd+++++++++ 2019-09-06_09-10-27/
2019/09/06 14:43:19 [3399] cd+++++++++ 2019-09-06_09-19-37/
2019/09/06 14:43:19 [3399] cd+++++++++ 2019-09-06_10-21-58/
2019/09/06 14:43:19 [3399] cd+++++++++ 2019-09-06_11-31-15/
2019/09/06 14:43:19 [3399] cd+++++++++ 2019-09-06_12-08-10/
2019/09/06 14:43:19 [3399] cd+++++++++ 2019-09-06_12-19-15/
2019/09/06 14:43:19 [3399] cd+++++++++ 2019-09-06_12-22-42/
2019/09/06 14:43:19 [3399] cd+++++++++ 2019-09-06_12-35-58/
2019/09/06 14:43:19 [3399] cd+++++++++ 2019-09-06_12-37-04/
2019/09/06 14:43:19 [3399] cd+++++++++ 2019-09-06_13-52-02/
2019/09/06 14:43:19 [3399] Number of files: 283 (reg: 273, dir: 10)
2019/09/06 14:43:19 [3399] Number of created files: 10 (dir: 10)
2019/09/06 14:43:19 [3399] Number of deleted files: 0
2019/09/06 14:43:19 [3399] Number of regular files transferred: 0
2019/09/06 14:43:19 [3399] Total file size: 7.61G bytes
2019/09/06 14:43:19 [3399] Total transferred file size: 0 bytes
2019/09/06 14:43:19 [3399] Literal data: 0 bytes
2019/09/06 14:43:19 [3399] Matched data: 0 bytes
2019/09/06 14:43:19 [3399] File list size: 0
2019/09/06 14:43:19 [3399] File list generation time: 0.005 seconds
2019/09/06 14:43:19 [3399] File list transfer time: 0.000 seconds
2019/09/06 14:43:19 [3399] Total bytes sent: 7.23K
2019/09/06 14:43:19 [3399] Total bytes received: 1.84K
2019/09/06 14:43:19 [3399] sent 7.23K bytes  received 1.84K bytes  6.04K bytes/sec
2019/09/06 14:43:19 [3399] total size is 7.61G  speedup is 839,114.58
2019/09/06 14:43:19 [3399] rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1196) [sender=3.1.2]
```

marcone, 2019-09-06:
That looks like like the problem is on the server/archive side, since it doesn't have permission to create the output folders there.

bensechrist, 2019-09-06:
Gotcha. My bad. Sync is working now

closed 2019-09-06 by bensechrist