marcone/teslausb #206: Design to 3D print a Pi Holder for Model 3

wowiejr, 2019-08-30:
[3D Printed Model 3 Pi Holder](https://www.thingiverse.com/thing:3837448)

Not really an issue but I wanted to share this with this community.

I designed this to hold and mount a Raspberry Pi Zero W in a Vilros case. Held in place by the usb's plugged into the car. Double sided tape could also be used.

[image]

Feel free to delete this if not appropriate.

Thanks again!

SemoTech, 2019-09-18:
Nice and creative @wowiejr, but you may want to just get a shorter cable :-) They are just $1-$2 on Amazon...

wowiejr, 2019-09-18:
Oh, for sure that is on the list. But I wanted to make it useful for multiple people, thus the wire guides/wraps.

SemoTech, 2019-09-18:
Yeah @wowiejr very creative and practical.  I actually separated the cavernous hold that is the front of the console in 2 layers.  At the bottom I have the Pi, a Zendue Pack, a Metromile plug, a HotSpot and a power distribution block, see picture [HERE](https://i.imgur.com/7obTYqs.jpg).  On top of that I installed a [full length tray](https://www.amazon.com/gp/product/B07TWHC41H/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) that hides everything and gives me space to put my daily stuff without having to dig for it while making everything neater. I am afraid that if I had that holder I could not fit the full top tray....

DineshCyanam, 2019-10-01:
@SemoTech: Do you have a guide on how to get additional power?

SemoTech, 2019-10-01:
I sure do @DineshCyanam Check my Modifications for 12V Power

[HERE](https://teslamotorsclub.com/tmc/posts/3933288/) 
and 
[HERE](https://teslamotorsclub.com/tmc/posts/3974968/)

miles267, 2019-10-01:
Is there a need for anything beyond a Pi Zero for this project if the SD card meets the spec?

SemoTech, 2019-10-01:
> Is there a need for anything beyond a Pi Zero for this project if the SD card meets the spec?

Nope!  The Pi 0 W works great with a good microSD card.  No errors and smooth recording.

closed 2019-10-01 by wowiejr