marcone/teslausb #210: How to properly test archive functionality without a car?

Helmi, 2019-08-31:
First of all thanks for taking the project and keeping it up to date. The one step setup worked great for me. I'm using a 128GB Samsung high endurance drive. Everything worked fine so far, the share appear and everything looks great only I'm not sure if archiving works and music copying doesn't seem to work although it might be the lack of mp3 files that causes problems.

What I did so far: I put two image files into the teslacam folder on the CAM share of the Pi and some m4a music files in my music share folder on the CIFS server (a synology ds). I then cut the pi from power, left it off for a few seconds and repowered it.

What happend is the image files got converted into .rec files and moved from Teslacam folder up to the root of the share. Nothing got copied to the cifs share setup for archiving. Also the music didn't get copied over.

I wonder if the files in TeslaCam need to follow a specific format and if the music folder also only is allowed to contain special formats . Or do I have to look elsewhere for a potential problem?

Thx,
Helmi

marcone, 2019-08-31:
Teslausb doesn't just archive any random file it finds. It only archives files that the car's Sentry or manual save put on the card, so you'll have to follow the naming scheme that the car uses.
For music syncing it does copy every file it finds in the music share.
The 'rec' files are files that the filesystem checker recovered.
You can perhaps find what went wrong by looking at the output of `/root/bin/setup-teslausb diagnose` (see also https://github.com/marcone/teslausb/issues/35, pinned at the top of the issues list)

closed 2019-08-31 by marcone

Helmi, 2019-08-31:
alright, got it, thanks. Any hint on the naming scheme tesla usb is looking for? 

For music I'll check the output of the setup.