marcone/teslausb #211: Failing to run most things - including setup-teslausb

bahree, 2019-08-31:
So, I have had this working for the last couple of months or so but recently started having some issues in the car, with it not regonizing the drive and have had to unplug and reboot the MCU multiple time. Finally decided to reflash and start again from scratch.

Now, the image was deployed successfully, and the pi is on wifi - I can ssh into it, but when I plug it into Windows it doesn't work and is not recognized or mounted.

When I ssh in and try and run setup-teslausb, it just hangs. I did a tail on /boot/teslausb-headless-setup.log and all I see if the following where a curl -s is failing. I can get to that URL, and see the json (pasted post the error messages) and it doesn't seem to be a firewall or network issue.

I cannot even generate a diagnostics, as it never returns anything and is stuck. As I said, this has worked before and I am using the same setup variable conf file, the only difference being I decided to add a music share.

And the timezone is also off and not getting update if that is any indication.

Appreciate any pointers or how I can help debug this.

Thanks.

```
Thu  3 Nov 15:51:23 PDT 2016 : 'curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev' failed, retrying
Thu  3 Nov 15:51:27 PDT 2016 : 'curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev' failed, retrying
Thu  3 Nov 15:51:30 PDT 2016 : 'curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev' failed, retrying
Thu  3 Nov 15:51:33 PDT 2016 : 'curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev' failed, retrying
Thu  3 Nov 15:51:37 PDT 2016 : 'curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev' failed, retrying
Thu  3 Nov 15:51:40 PDT 2016 : 'curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev' failed, retrying
Thu  3 Nov 15:51:44 PDT 2016 : 'curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev' failed, retrying
Thu  3 Nov 15:51:47 PDT 2016 : 'curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev' failed, retrying
Thu  3 Nov 15:51:50 PDT 2016 : 'curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev' failed, retrying
Thu  3 Nov 15:51:54 PDT 2016 : 'curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev' failed, retrying
Thu  3 Nov 15:51:57 PDT 2016 : 'curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev' failed, retrying
Thu  3 Nov 15:52:00 PDT 2016 : 'curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev' failed, retrying
Thu  3 Nov 15:52:04 PDT 2016 : 'curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev' failed, retrying
Thu  3 Nov 15:52:07 PDT 2016 : 'curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev' failed, retrying
Thu  3 Nov 15:52:10 PDT 2016 : 'curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev' failed, retrying
Thu  3 Nov 15:52:14 PDT 2016 : 'curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev' failed, retrying
```

And here is the json from manually browsing to it.
```
{
  "ref": "refs/heads/main-dev",
  "node_id": "MDM6UmVmMTczNjA2MzEwOm1haW4tZGV2",
  "url": "https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev",
  "object": {
    "sha": "6d0d649be57ca14d997774fbc1a255454fe9c8c1",
    "type": "commit",
    "url": "https://api.github.com/repos/marcone/teslausb/git/commits/6d0d649be57ca14d997774fbc1a255454fe9c8c1"
  }
}
```

marcone, 2019-09-01:
Anything in archiveloop.log?

bahree, 2019-09-01:
Thanks @marcone . Here is what I see in that - dumping the full log here, in case the sequence helps but seems a repeat. Any ideas?

```
Thu  3 Nov 10:17:07 PDT 2016: Starting archiveloop...
Thu  3 Nov 10:17:07 PDT 2016: no snapshots
Thu  3 Nov 10:17:07 PDT 2016: mounted all snapshots
Thu  3 Nov 10:17:08 PDT 2016: taking snapshot of cam disk: /backingfiles/snapshots/snap-000000/snap.bin
Thu  3 Nov 10:17:11 PDT 2016: took snapshot
Thu  3 Nov 10:17:11 PDT 2016: comparing /backingfiles/snapshots/snap--00001/snap.bin.toc and /backingfiles/snapshots/newsnap/snap.bin.toc
Thu  3 Nov 10:17:11 PDT 2016: making links for /backingfiles/snapshots/newsnap/mnt, retargeted to /backingfiles/snapshots/snap-000000/mnt
Thu  3 Nov 10:17:11 PDT 2016: made all links for /backingfiles/snapshots/newsnap/mnt
Thu  3 Nov 10:17:11 PDT 2016: Waiting for time to be set by ntpd...
Thu  3 Nov 10:17:37 PDT 2016: Time still not set, attempting to force it
Thu  3 Nov 11:15:31 PDT 2016: taking snapshot of cam disk: /backingfiles/snapshots/snap-000001/snap.bin
Thu  3 Nov 11:15:33 PDT 2016: took snapshot
Thu  3 Nov 11:15:33 PDT 2016: comparing /backingfiles/snapshots/snap-000000/snap.bin.toc and /backingfiles/snapshots/newsnap/snap.bin.toc
Thu  3 Nov 11:15:33 PDT 2016: new snapshot is identical to previous one, discarding
Thu  3 Nov 12:13:56 PDT 2016: taking snapshot of cam disk: /backingfiles/snapshots/snap-000001/snap.bin
Thu  3 Nov 12:13:58 PDT 2016: took snapshot
Thu  3 Nov 12:13:58 PDT 2016: comparing /backingfiles/snapshots/snap-000000/snap.bin.toc and /backingfiles/snapshots/newsnap/snap.bin.toc
Thu  3 Nov 12:13:58 PDT 2016: new snapshot is identical to previous one, discarding
==============================================
Thu  3 Nov 10:17:06 PDT 2016: Starting archiveloop...
Thu  3 Nov 10:17:08 PDT 2016: mounted all snapshots
Thu  3 Nov 10:17:09 PDT 2016: taking snapshot of cam disk: /backingfiles/snapshots/snap-000001/snap.bin
Thu  3 Nov 10:17:11 PDT 2016: took snapshot
Thu  3 Nov 10:17:11 PDT 2016: comparing /backingfiles/snapshots/snap-000000/snap.bin.toc and /backingfiles/snapshots/newsnap/snap.bin.toc
Thu  3 Nov 10:17:11 PDT 2016: new snapshot is identical to previous one, discarding
Thu  3 Nov 10:17:14 PDT 2016: Waiting for time to be set by ntpd...
Thu  3 Nov 10:17:40 PDT 2016: Time still not set, attempting to force it
Thu  3 Nov 11:15:34 PDT 2016: taking snapshot of cam disk: /backingfiles/snapshots/snap-000001/snap.bin
Thu  3 Nov 11:15:36 PDT 2016: took snapshot
Thu  3 Nov 11:15:36 PDT 2016: comparing /backingfiles/snapshots/snap-000000/snap.bin.toc and /backingfiles/snapshots/newsnap/snap.bin.toc
Thu  3 Nov 11:15:36 PDT 2016: new snapshot is identical to previous one, discarding
Thu  3 Nov 12:13:59 PDT 2016: taking snapshot of cam disk: /backingfiles/snapshots/snap-000001/snap.bin
Thu  3 Nov 12:14:01 PDT 2016: took snapshot
Thu  3 Nov 12:14:01 PDT 2016: comparing /backingfiles/snapshots/snap-000000/snap.bin.toc and /backingfiles/snapshots/newsnap/snap.bin.toc
Thu  3 Nov 12:14:01 PDT 2016: new snapshot is identical to previous one, discarding
Thu  3 Nov 13:12:24 PDT 2016: taking snapshot of cam disk: /backingfiles/snapshots/snap-000001/snap.bin
Thu  3 Nov 13:12:27 PDT 2016: took snapshot
Thu  3 Nov 13:12:27 PDT 2016: comparing /backingfiles/snapshots/snap-000000/snap.bin.toc and /backingfiles/snapshots/newsnap/snap.bin.toc
Thu  3 Nov 13:12:27 PDT 2016: new snapshot is identical to previous one, discarding
Thu  3 Nov 14:10:49 PDT 2016: taking snapshot of cam disk: /backingfiles/snapshots/snap-000001/snap.bin
Thu  3 Nov 14:10:52 PDT 2016: took snapshot
Thu  3 Nov 14:10:52 PDT 2016: comparing /backingfiles/snapshots/snap-000000/snap.bin.toc and /backingfiles/snapshots/newsnap/snap.bin.toc
Thu  3 Nov 14:10:52 PDT 2016: new snapshot is identical to previous one, discarding
Thu  3 Nov 15:09:14 PDT 2016: taking snapshot of cam disk: /backingfiles/snapshots/snap-000001/snap.bin
Thu  3 Nov 15:09:17 PDT 2016: took snapshot
Thu  3 Nov 15:09:17 PDT 2016: comparing /backingfiles/snapshots/snap-000000/snap.bin.toc and /backingfiles/snapshots/newsnap/snap.bin.toc
Thu  3 Nov 15:09:17 PDT 2016: new snapshot is identical to previous one, discarding
Thu  3 Nov 16:07:40 PDT 2016: taking snapshot of cam disk: /backingfiles/snapshots/snap-000001/snap.bin
Thu  3 Nov 16:07:42 PDT 2016: took snapshot
Thu  3 Nov 16:07:42 PDT 2016: comparing /backingfiles/snapshots/snap-000000/snap.bin.toc and /backingfiles/snapshots/newsnap/snap.bin.toc
Thu  3 Nov 16:07:42 PDT 2016: new snapshot is identical to previous one, discarding
Thu  3 Nov 17:06:05 PDT 2016: taking snapshot of cam disk: /backingfiles/snapshots/snap-000001/snap.bin
Thu  3 Nov 17:06:07 PDT 2016: took snapshot
Thu  3 Nov 17:06:07 PDT 2016: comparing /backingfiles/snapshots/snap-000000/snap.bin.toc and /backingfiles/snapshots/newsnap/snap.bin.toc
Thu  3 Nov 17:06:07 PDT 2016: new snapshot is identical to previous one, discarding
==============================================
Thu  3 Nov 10:17:06 PDT 2016: Starting archiveloop...
Thu  3 Nov 10:17:08 PDT 2016: mounted all snapshots
Thu  3 Nov 10:17:09 PDT 2016: taking snapshot of cam disk: /backingfiles/snapshots/snap-000001/snap.bin
Thu  3 Nov 10:17:11 PDT 2016: took snapshot
Thu  3 Nov 10:17:11 PDT 2016: comparing /backingfiles/snapshots/snap-000000/snap.bin.toc and /backingfiles/snapshots/newsnap/snap.bin.toc
Thu  3 Nov 10:17:11 PDT 2016: new snapshot is identical to previous one, discarding
[... cut ...]
Thu  3 Nov 13:12:26 PDT 2016: comparing /backingfiles/snapshots/snap-000000/snap.bin.toc and /backingfiles/snapshots/newsnap/snap.bin.toc
Thu  3 Nov 13:12:26 PDT 2016: new snapshot is identical to previous one, discarding
Thu  3 Nov 14:10:49 PDT 2016: taking snapshot of cam disk: /backingfiles/snapshots/snap-000001/snap.bin
Thu  3 Nov 14:10:51 PDT 2016: took snapshot
Thu  3 Nov 14:10:51 PDT 2016: comparing /backingfiles/snapshots/snap-000000/snap.bin.toc and /backingfiles/snapshots/newsnap/snap.bin.toc
Thu  3 Nov 14:10:51 PDT 2016: new snapshot is identical to previous one, discarding
```

marcone, 2019-09-01:
You say you can ssh into the Pi, but the archiveloop log never shows it connecting to the archive server, nor did ntp adjust the date/time.
So I'd start by looking at your network config and see if there's anything wrong there.

bahree, 2019-09-01:
I agree, but at the same time this is already on the wifi (which is how I can SSH in and also manually run that curl statement). I changed SD cards too just to make sure. The only thing I can think of now is that it might be something with the pi itself, or the wifi radio. Don't have a 2nd one and will need to order one. Any ideas if this will also work with a regular Raspberry Pi 3 or only a Zero W? Thanks.

marcone, 2019-09-02:
Can you ping the archive server or other machines manually? If you're ssh-ing in over wifi then clearly the wifi is working, so it might a routing or DNS problem.

The Pi 3 cannot act as a USB gadget, so teslausb won't work with that, but supposedly the Pi 4B can. I don't have a 4B to confirm that though.

bahree, 2019-09-03:
Yes, I can ping the server and a number of machines on the network. I don't know if this is a wonky hardware issue or something else. I got a new pi ordered and will try that and report back here. And also got a new Pi4 and can try that too and share here. Thanks for your help!

bahree, 2019-09-09:
OK a couple of updates - got another raspberry PI zero and different set of cards and I can replicate the issues. I was running a UTM at home with a SSL inspector which might be the issue in updating (even though my rules are setup to allow the pi's to go thru without any issues). Irrespective, remove that and eliminated the issues. Also went back and added the certs to ensure this doens't happen again.

So from this all I can run the tesla-update and everything is upto date. 

However the one issue is that the drive doesn't mount the archive - even when it is configured and setup. When I cd to /mnt/archive and do a ls, I don't see anything.

I have to run mount -a again, and then it mounts and shows up - I don't know why. I haven't changed the fstab file.

The diagnostics [output is pasted here](https://pastebin.com/q4F1X41s). Appreciate any insight.

Thanks.

marcone, 2019-09-09:
The archive script unmounts the archive share after the archive operation is complete (and then remounts it when it is needed for the next archive operation). Could it be you sshd in after archiving finished?

fat-tire, 2019-09-14:
I don't know if this is the same issue, but I was seeing a hang as well.  The feedback to the user for `setup-teslausb` is just awful.  You should at least know what it's TRYING to do.

The problem in my case was that the certificate for `https://api.github.com` as well as in `raspbian.raspberrypi.org/raspbian/dists/buster/InRelease` was 'not yet valid' because NTP on the pi is totally useless and so it failed silently.  Make sure you're in read-write mode & sudo -i and try:

     # service ntp stop
     # ntpd -q -g
     # service ntp start

to set the date manually to something that sort of makes sense.

Now it should work.   ntp needs to get fixed tho.

marcone, 2019-10-24:
the current archiveloop stops and starts ntp to force time to be set

closed 2019-10-24 by marcone