marcone/teslausb #212: Not archiving.. "Device or resource busy"

spalt, 2019-09-01:
I'm also in the boat where it used to work great but lately I'm getting some weird issues and clips are not archiving..  archiveloop.log:

==============================================
Sun  1 Sep 08:44:24 EDT 2019: Starting archiveloop...
Sun  1 Sep 08:44:25 EDT 2019: Ensuring cam file is mounted...
Sun  1 Sep 08:44:25 EDT 2019: Mounting /mnt/cam...
Sun  1 Sep 08:44:25 EDT 2019: Mounted /mnt/cam.
Sun  1 Sep 08:44:25 EDT 2019: Ensured cam file is mounted.
Sun  1 Sep 08:44:25 EDT 2019: Running fsck on /mnt/cam...
Sun  1 Sep 08:44:25 EDT 2019: | fsck from util-linux 2.29.2
Sun  1 Sep 08:44:26 EDT 2019: | fsck.fat 4.1 (2017-01-24)
Sun  1 Sep 08:44:26 EDT 2019: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Sun  1 Sep 08:44:26 EDT 2019: |  Automatically removing dirty bit.
Sun  1 Sep 08:44:26 EDT 2019: | Performing changes.
Sun  1 Sep 08:44:26 EDT 2019: | /dev/loop0: 203 files, 1009/1397614 clusters
Sun  1 Sep 08:44:26 EDT 2019: Finished fsck on /mnt/cam.
Sun  1 Sep 08:44:26 EDT 2019: Unmounting /mnt/cam...
Sun  1 Sep 08:44:27 EDT 2019: Unmounted /mnt/cam.
Sun  1 Sep 08:44:27 EDT 2019: Ensuring music backing file is mounted...
Sun  1 Sep 08:44:27 EDT 2019: Mounting /mnt/music...
Sun  1 Sep 08:44:27 EDT 2019: Mounted /mnt/music.
Sun  1 Sep 08:44:27 EDT 2019: Ensured music drive is mounted.
Sun  1 Sep 08:44:27 EDT 2019: Running fsck on /mnt/music...
Sun  1 Sep 08:44:27 EDT 2019: | fsck from util-linux 2.29.2
Sun  1 Sep 08:44:28 EDT 2019: | fsck.fat 4.1 (2017-01-24)
Sun  1 Sep 08:44:28 EDT 2019: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Sun  1 Sep 08:44:28 EDT 2019: |  Automatically removing dirty bit.
Sun  1 Sep 08:44:28 EDT 2019: | Performing changes.
Sun  1 Sep 08:44:28 EDT 2019: | /dev/loop0: 40 files, 3973/1048754 clustersSun  1 Sep 08:44:28 EDT 2019: Finished fsck on /mnt/music.
Sun  1 Sep 08:44:28 EDT 2019: Unmounting /mnt/music...
Sun  1 Sep 08:44:29 EDT 2019: Unmounted /mnt/music.
Sun  1 Sep 08:44:29 EDT 2019: Connecting usb to host...
Sun  1 Sep 08:44:29 EDT 2019: Connected usb to host.
Sun  1 Sep 08:44:29 EDT 2019: Waiting for archive to be reachable...
Sun  1 Sep 08:44:29 EDT 2019: Archive is reachable.
Sun  1 Sep 08:44:29 EDT 2019: Waiting for time to be set by ntpd...
Sun  1 Sep 08:44:30 EDT 2019: Time now set
Sun  1 Sep 08:44:50 EDT 2019: Archiving...
Sun  1 Sep 08:44:50 EDT 2019: Disconnecting usb from host...
Sun  1 Sep 08:44:50 EDT 2019: Disconnected usb from host.
Sun  1 Sep 08:44:50 EDT 2019: Ensuring cam archive is mounted...
Sun  1 Sep 08:44:50 EDT 2019: /mnt/archive is already mounted.
Sun  1 Sep 08:44:50 EDT 2019: Ensured cam archive is mounted.
Sun  1 Sep 08:44:50 EDT 2019: Ensuring music archive is mounted...
Sun  1 Sep 08:44:50 EDT 2019: /mnt/musicarchive is already mounted.
Sun  1 Sep 08:44:50 EDT 2019: Ensured music archive is mounted.
Sun  1 Sep 08:44:50 EDT 2019: Starting recording archiving...
Sun  1 Sep 08:44:50 EDT 2019: Sending IFTTT message for moved files.
Sun  1 Sep 08:44:51 EDT 2019: Ensuring cam file is mounted...
Sun  1 Sep 08:44:51 EDT 2019: Mounting /mnt/cam...
Sun  1 Sep 08:44:51 EDT 2019: Mounted /mnt/cam.
Sun  1 Sep 08:44:51 EDT 2019: Ensured cam file is mounted.
Sun  1 Sep 08:44:51 EDT 2019: Running fsck on /mnt/cam...
Sun  1 Sep 08:44:51 EDT 2019: | fsck from util-linux 2.29.2
Sun  1 Sep 08:44:52 EDT 2019: | fsck.fat 4.1 (2017-01-24)
Sun  1 Sep 08:44:52 EDT 2019: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Sun  1 Sep 08:44:52 EDT 2019: |  Automatically removing dirty bit.
Sun  1 Sep 08:44:52 EDT 2019: | Performing changes.
Sun  1 Sep 08:44:52 EDT 2019: | /dev/loop0: 203 files, 1009/1397614 clusters
Sun  1 Sep 08:44:52 EDT 2019: Finished fsck on /mnt/cam.
Sun  1 Sep 08:44:52 EDT 2019: wake up car process is 1342
Sun  1 Sep 08:44:52 EDT 2019: Moving clips to archive...
Sun  1 Sep 08:44:52 EDT 2019: Creating output directory ''
Sun  1 Sep 08:44:52 EDT 2019: Creating output directory ''
Sun  1 Sep 08:44:52 EDT 2019: Moved 0 file(s), failed to copy 0, deleted 0.Sun  1 Sep 08:44:52 EDT 2019: Finished moving clips to archive.
Sun  1 Sep 08:44:52 EDT 2019: killing PID 1342
Sun  1 Sep 08:44:52 EDT 2019: Unmounting /mnt/cam...
Sun  1 Sep 08:44:53 EDT 2019: Unmounted /mnt/cam.
Sun  1 Sep 08:44:53 EDT 2019: Finished archiving.
Sun  1 Sep 08:44:53 EDT 2019: Copying music...
Sun  1 Sep 08:44:53 EDT 2019: Starting music sync...
Sun  1 Sep 08:44:53 EDT 2019: Ensuring music backing file is mounted...
Sun  1 Sep 08:44:53 EDT 2019: Mounting /mnt/music...
Sun  1 Sep 08:44:53 EDT 2019: Mounted /mnt/music.
Sun  1 Sep 08:44:53 EDT 2019: Ensured music drive is mounted.
Sun  1 Sep 08:44:53 EDT 2019: Running fsck on /mnt/music...
Sun  1 Sep 08:44:53 EDT 2019: | fsck from util-linux 2.29.2
Sun  1 Sep 08:44:54 EDT 2019: | fsck.fat 4.1 (2017-01-24)
Sun  1 Sep 08:44:54 EDT 2019: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Sun  1 Sep 08:44:54 EDT 2019: |  Automatically removing dirty bit.
Sun  1 Sep 08:44:54 EDT 2019: | Performing changes.
Sun  1 Sep 08:44:54 EDT 2019: | /dev/loop0: 40 files, 3973/1048754 clustersSun  1 Sep 08:44:54 EDT 2019: Finished fsck on /mnt/music.
Sun  1 Sep 08:44:54 EDT 2019: Copying music from archive...
Sun  1 Sep 08:44:54 EDT 2019: Copied 0 music file(s), skipped 28 previously-copied files, encountered 0 errors.
Sun  1 Sep 08:44:54 EDT 2019: Unmounting /mnt/music...
Sun  1 Sep 08:44:54 EDT 2019: Unmounted /mnt/music.
Sun  1 Sep 08:44:54 EDT 2019: Finished copying music.
Sun  1 Sep 08:44:54 EDT 2019: unmounting /mnt/archive
Sun  1 Sep 08:44:54 EDT 2019: unmounting /mnt/musicarchive
Sun  1 Sep 08:44:55 EDT 2019: Connecting usb to host...
Sun  1 Sep 08:44:55 EDT 2019: Connected usb to host.
Sun  1 Sep 08:44:55 EDT 2019: Waiting for archive to be unreachable...
pi@teslausb:~ $

_Originally posted by @spalt in https://github.com/marcone/teslausb/issues/187#issuecomment-526918157_

Running ./archiveloop manually, I get this:

root@teslausb:~/bin# ./archiveloop
Powering off HDMI
rm: cannot remove '/backingfiles/snapshots/newsnap/mnt': Device or resource busy
/dev/loop2p1 on /backingfiles/snapshots/newsnap/mnt type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
snapshot already mounted
fsck from util-linux 2.29.2
fsck.fat 4.1 (2017-01-24)
/dev/loop3p1: 203 files, 1009/1397614 clusters
stat: cannot stat '/backingfiles/snapshots/newsnap/mnt/TeslaCam/SavedClips/*/*': No such file or directory
mv: cannot move '/backingfiles/snapshots/newsnap' to '/backingfiles/snapshots/snap-000000/newsnap': File exists
find: ‘/mnt/cam/TeslaCam/SavedClips/*’: No such file or directory
find: ‘/mnt/cam/TeslaCam/SavedClips/*’: No such file or directory
rm: cannot remove '/backingfiles/snapshots/newsnap/mnt': Device or resource busy
/dev/loop2p1 on /backingfiles/snapshots/newsnap/mnt type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/loop3p1 on /backingfiles/snapshots/newsnap/mnt type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
snapshot already mounted
fsck from util-linux 2.29.2
fsck.fat 4.1 (2017-01-24)
/dev/loop4p1: 203 files, 1009/1397614 clusters
stat: cannot stat '/backingfiles/snapshots/newsnap/mnt/TeslaCam/SavedClips/*/*': No such file or directory
mv: cannot move '/backingfiles/snapshots/newsnap' to '/backingfiles/snapshots/snap-000000/newsnap': File exists

then it just repeats....

marcone, 2019-09-01:
The archiveloop log indicates there was nothing to archive.
The "device or resource busy" seems to indicate you killed the archiveloop while it was in the middle of taking a snapshot.
The 595 byte files are a problem on the Tesla side, usually fixed by rebooting the car. Since these files are unplayable, they are never archived.

closed 2019-09-01 by marcone

spalt, 2019-09-01:
ok thank you.  i deleted the 595 byte files in /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips, rebooted the car, and will see if that makes any difference...