marcone/teslausb #214: Reboot loop when connected to vehicle by single cable

rohan3au, 2019-09-02:
I've got the Pi working perfectly when plugged into my computer, both drives show up, it syncs with the network share perfectly.

When connecting to my vehicle, it gets stuck in a reboot loop (2 flashes, thinks for 15 seconds, and repeat). If I connect to the power only USB port on the Pi, it boots fine and THEN connect also the data/power USB port to the second USB port on the car, everything works just fine, but this has to be done every single time the car is turned off (turns the USB ports off), so it's not exactly convenient and set-and-forget.

It seems like the car is cutting the power to the Pi after 15 seconds or so then retrying, I have a feeling this may be due to my vehicle, which is one of only a small set of vehicles between Sept '17 and March '18 that is MCU 1 (Nvidia Tegra 3 ARM Processor, rather than the Intel processors in MCU2), but also has AP2.5 (so has dash cam and sentry mode).

Any advice? Thoughts?

marcone, 2019-09-02:
Unfortunately this seems to be a problem with some cars, and all models are apparently affected.
See issue https://github.com/marcone/teslausb/issues/9

closed 2019-09-02 by marcone

jun3280net, 2019-11-08:
> I've got the Pi working perfectly when plugged into my computer, both drives show up, it syncs with the network share perfectly.
>
> When connecting to my vehicle, it gets stuck in a reboot loop (2 [...]

Same vehicle as you - and experiencing the same problems.  Definitely seems to be related to the vehicle.  

I tested by setting up the PI from my computer USB to PI USB port (not Power port).  PI was able to install, reboot, and startup without getting stuck in the bootloop.

However in the Model S (Tegra/AP2.5), I have to connect to Power port first, wait around 3 minutes till it blinks like a heartbeat (2 beats), then plug the USB port to Tesla USB.

As @marcone indicated - I think a battery portable backup is the best solution for us.  I'm not sure how or where he taped his USB cord to convert it to a data only cable (I'll have to Google that).

Cheers,

J

jun3280net, 2019-11-08:
@rohan3au and @marcone  - not sure if either of you tried this, but I was reading another thread specific to MCU1/AP2.5, and @awisur suggested using a USB Hub to connect between the PI USB port and Tesla USB port for his issue.  

I don't understand how - but this solved the bootloop problem for me.

For reference, I used this Anker USB Hub I got from Amazon.

https://www.amazon.ca/Anker-Unibody-Aluminum-Gigabit-Ethernet/dp/B00PC07T02/ref=sr_1_2?crid=2GI0A2PE046C0&keywords=anker+aluminum+unibody+gigabit+usb-c&qid=1573230403&sprefix=anker+usb+gig%2Caps%2C234&sr=8-2

[image]