marcone/teslausb #215: Can't seem to get it to recognize the cifs share

bobk2824, 2019-09-02:
I've re-flashed this a number of times and it always ends up with the LED off, no matter how long I wait.
Then I unplug the PI and plug it in, and it gets pretty far, but ends with the repeated 2 flashes.

When I ssh in, I see:
WIFI_ENABLED
TESLAUSB_SETUP_FINISHED

In looking at teslausb-headless-setup.log (attached) I see lots of these:
Mon  2 Sep 11:17:06 EDT 2019 : verify-and-configure-archive: Trying mount command-line:
Mon  2 Sep 11:17:06 EDT 2019 : verify-and-configure-archive: mount -t cifs '//bobdev/teslacam' '/tmp/archivetestmount' -o 'credentials=/tmp/teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777,,'

And eventually this:
Mon  2 Sep 11:17:11 EDT 2019 : verify-and-configure-archive: STOP: no working combination of vers and sec mount options worked

If I then type in the mount command manually, it mounts it with no problem. Wifi is working great.

If I run /etc/rc.local, it says: already running.

Any idea on how I can get it to retry the mount? Unfortunately, reflashing doesn't help.

Thanks,
Bob

[teslausb-headless-setup.log](https://github.com/marcone/teslausb/files/3566606/teslausb-headless-setup.log)

marcone, 2019-09-02:
You can run /root/bin/setup-teslausb to rerun setup.
If there's a specific combination of options that works, you can try if adding those options to the .conf file will make it work (if cifs_version and cifs_sec are set, it will try only those options).

bobk2824, 2019-09-02:
Okay! Running it that way I was able to see the credentials file it presents to mount and I see what the problem is. In the .conf file there is no variable for the 'sharedomain', so I put this:
export sharename=teslacam
export shareuser=domain\user
export sharepassword=password

Being windows-centric, I did this basically without thinking.

The script can't handle that. I changed it to be just the user and it works well.

Thanks for your help!
Bob

closed 2019-09-02 by bobk2824