marcone/teslausb #216: Successful setup / car is recording recent clips / but no Camera icon showing up in car?

johnny2678, 2019-09-02:
Hi, first off - thanks for this.  Really cool project!

Did a search through the issues and didn't see mine.  Hope I'm not missing something obvious.

Issue: The display doesn't show the Camera icon (with or without red dot).

I got through the setup just fine. rPI can see my CIFS server and I can SSH to the rPI when it's connected to the car.

After a quick trip around the block, I do have clips in my Recent clips folder:
```
pi@teslausb:/backingfiles/TeslaCam/RecentClips $ ls
2019-09-02_15-49-17-front.mp4           2019-09-02_15-50-18-front.mp4           2019-09-02_15-51-18-front.mp4
2019-09-02_15-49-17-left_repeater.mp4   2019-09-02_15-50-18-left_repeater.mp4   2019-09-02_15-51-18-left_repeater.mp4
2019-09-02_15-49-17-right_repeater.mp4  2019-09-02_15-50-18-right_repeater.mp4  2019-09-02_15-51-18-right_repeater.mp4
```

So it looks like it's recording just fine, concern is I can't manually add anything to the SavedClips folder by tapping the camera icon.

Any suggestions?

DMESG: https://pastebin.com/uSdGTT8f
tslausb-headless-setup.log: https://pastebin.com/gCQprXBQ
archiveloop.log: https://pastebin.com/ZeDJeZ1H
teslausb_setup_variables.conf (redacted): https://pastebin.com/fGwjNSDi

DMBlakeley, 2019-09-02:
As problem appears to be on the Tesla side, I would suggest rebooting your Tesla. On M3 you hold both left and right steering wheel buttons in until the Tesla “T” shows on the screen and then release.

marcone, 2019-09-03:
It does seem like the car recording things while not showing the dashcam icon is a car problem, not a teslausb problem.

johnny2678, 2019-09-03:
hmm... car is down for the night ;)

I'll give it another shot in the AM.

marcone, 2019-09-04:
@johnny2678  Any luck?

johnny2678, 2019-09-04:
Hi @marcone... thx for checking in.  I haven't posted because I'm not really sure how to explain what I'm seeing. I was going to complete a few more drives to try and make sense of it.

On my latest drive, Teslausb only started working after i did a soft reboot of the MCU.  Once i did, the camera icon showed up and even synced clips that I saved to my local nas (which is really cool, btw).

But then I let the car sit and fired it up again and the camera icon is gone.  I might just have to reboot the MCU on every drive, which I can live with.

Was going to complete a few more drives

tomstacey, 2019-09-04:
I have the same issue with the camera icon not showing up.. Will try a reboot too. Should the TeslaUSB mount as a drive if I plug it into my Mac and let it boot? (as it doesn't)

marcone, 2019-09-04:
@tomstacey Yes, the Mac should see a "CAM" disk, and if you configured space for music also a MUSIC disk.
Most common cause of it not working is using a charge-only cable.

@johnny2678 Seems like an issue on the car side of things then, so I'm going to close this issue.

closed 2019-09-04 by marcone

tomstacey, 2019-09-05:
It doesn't mount on my Mac... I have re-imaged and have the same issue. I have tried lots of cables now. Anything else to try?

marcone, 2019-09-05:
Make sure you plug them into the right port? 😄
More seriously, some troubleshooting steps are in the wiki, maybe something in there will help (https://github.com/marcone/teslausb/wiki).

johnny2678, 2019-09-05:
HI @marcone - only thing holding me back is the camera w/ red dot shows up every time on my M3 when a normal Sandisk usb stick is plugged in.

For some reason, the car is powering the rPi when I get in the car, but the icon doesn't show up until I do an MCU reboot.  At this point, I have confirmed with multiple drives from cold boots.

Maybe a boot / mount timing thing?  Need to dig into the code a little bit to see what's happening.

I'll post back here with any observations. If you have any pointers on any parameters I could tweak and try, let me know.

tomstacey, 2019-09-05:
OH! Of course! As my Pi is a Zero (non W), I have an OTG cable with a USB Wifi NIC in it.  Any way I can make it work on my Pi Zero?

Sorry for not RTFM :-)

marcone, 2019-09-05:
It has to be a Pi Zero W.
Apparently a Pi 4B will also work, I haven't tested it.

tomstacey, 2019-09-05:
I have a Pi 4B…! (And a Pi 1 B)

I’m wondering if I can do anything with a Hub?

johnny2678, 2019-09-08:
just to close the loop on the original issue - I re-flashed and everything seems to be working now. Not sure if it makes a difference but I changed Cam folder size from 32G to 100% (on a 128GB card).