marcone/teslausb #219: Coding Questions

DonC52, 2019-09-03:
If I have coding questions, where is the best place to get those asked. Issues does not seem like the place to do that.

marcone, 2019-09-03:
stackoverflow?

DonC52, 2019-09-03:
No - just looking through code and wondered how stuff is actually working.

An example is line 46 of the archive-clips.sh file, 
It is the "while read file_name" in the function moveclips()  Where is the input for that read supplied? 

function moveclips() {
  ROOT="$1"
  PATTERN="$2"
  if [ ! -d "$ROOT" ]
  then
    log "$ROOT does not exist, skipping"
    return
  fi
  while read file_name
  do
....

The function call does not seem to have that.

moveclips "$CAM_MOUNT/TeslaCam/SavedClips" '*'

Coding stuff like that. I am trying to remember a lot of shell coding tricks, but that stumped me.

DonC52, 2019-09-03:
Also - is there a source .zip file with all of the updates that I seem to got during the install.  I got to learn by taking the Raspberry out of the car when I want to learn.

As well - I notice that the Pi does not always have the correct time and date. I dont think it matters as the files are date/time stamped in their names.  If I take a long trip and drive a lot from home, it may have a lot of security clips to save. Just want to be sure that the internal clock will not mess that up.

Is there any documents on architecture as well - what do the snapshots do other than have a pile of links. ;-)

This is a great project. Thanks for keeping it alive and well.

marcone, 2019-09-03:
> No - just looking through code and wondered how stuff is actually working.
> 
> An example is line 46 of the archive-clips.sh file,
> It is the "while read file_name" in the function moveclips() Where is the input for that read supplied?

It's at the end of the loop:
```
  done <<< $(cd "$ROOT"; find $PATTERN)
```

marcone, 2019-09-03:
> Also - is there a source .zip file with all of the updates that I seem to got during the install. 

Not sure what you mean.
 
> As well - I notice that the Pi does not always have the correct time and date. I dont think it matters as the files are date/time stamped in their names. If I take a long trip and drive a lot from [...]

It shouldn't matter.
 
> Is there any documents on architecture as well - what do the snapshots do other than have a pile of links. ;-)

The snapshots are used to save more than one hour of recording history. There is some description of the general architecture of teslausb in the main readme, but not for the snapshots or other improvements made since forking off the original project.

DonC52, 2019-09-03:
How could I miss that. It's right in front of my eyes. Thanks. 

So the snapshots are only the running videos (Recent) and not the SavedClips?

I noticed in another event that there is no real easy way to get the expanded snapshots (and what's required) installed. Is it simply a re-install? (That's not all that hard.)

On the source - you all have changed/added a bunch of stuff that is not in the source .zip file available for the project. I am new to github so I am not sure where the current sources are that correspond to what I am actually running.

marcone, 2019-09-03:
> How could I miss that. It's right in front of my eyes. Thanks.
> 
> So the snapshots are only the running videos (Recent) and not the SavedClips?

Currently yes, but I'll be adding SavedClips soon.

> I noticed in another event that there is no real easy way to get the expanded snapshots (and what's required) installed. Is it simply a re-install? (That's not all that hard.)

Reflash and setup from scratch is easy. It's a little harder to change an existing install, so I don't recommend that for people not familiar with teslausb.

> On the source - you all have changed/added a bunch of stuff that is not in the source .zip file available for the project. I am new to github so I am not sure where the current sources are that correspond to what I am actually running.

Setup always downloads the latest sources from Github at install time. You can use git to clone a local repository, or use the green "Clone or download" button on https://github.com/marcone/teslausb to download a zip file.

closed 2019-09-03 by marcone