marcone/teslausb #220: low space for snapshots

curiousgrge, 2019-09-03:
So I recently upgraded from teslausb-20190615 build on August 30th using teslausb selfupdate feature and right after this happened, I haven't been able to archive my recordings.

I spotted this in diagnostics.txt

Fri 30 Aug 01:33:29 PDT 2019: low space, deleting /backingfiles/snapshots/snap-000000
Fri 30 Aug 01:33:32 PDT 2019: warning: low space for snapshots
Fri 30 Aug 01:33:32 PDT 2019: taking snapshot of cam disk: /backingfiles/snapshots/snap-000000/snap.bin

My microSD card is a 128GB partitioned half for CAM and half for MUSIC.
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       1.6G  1.3G  228M  86% /
devtmpfs        213M     0  213M   0% /dev
tmpfs           217M     0  217M   0% /dev/shm
tmpfs           217M  5.9M  211M   3% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           217M     0  217M   0% /sys/fs/cgroup
tmpfs           217M   20K  217M   1% /var/log
tmpfs           217M  420K  216M   1% /var/cache/samba
tmpfs           217M  8.0K  217M   1% /tmp
tmpfs           217M     0  217M   0% /mnt/smbexport
tmpfs           217M     0  217M   0% /var/tmp
tmpfs           217M  176K  217M   1% /run/samba
/dev/mmcblk0p1   44M   23M   22M  52% /boot
/dev/mmcblk0p4   94M  5.7M   82M   7% /mutable
/dev/mmcblk0p3  118G  117G  1.0G 100% /backingfiles
/dev/loop0p1     58G  2.0G   56G   4% /backingfiles/snapshots/snap-000000/mnt
tmpfs            44M     0   44M   0% /run/user/1000

Not sure if snap.bin took over completely cam.disk because cam.disk is still taking up the exact same amount of space and would explain why it runs out of space.

I also noticed in a previous message that smb-on.sh and smb-off.sh is no longer available but it is still in my /root/bin folder.  Those are the only two files in that folder that still has my original install date.

Is the cleanup process not removing files when it should such as cam.disk, smb*.sh?

marcone, 2019-09-03:
What do you mean by "I haven't been able to archive my recordings" ?
What actually goes wrong? (please see https://github.com/marcone/teslausb/issues/35 for how to post the required information)

curiousgrge, 2019-09-03:
Archive as in being able to sync the saved files onto share destination.  Prior to the update, the archive process would process files and move them to the share then my notification program, Pushover, would send me a summary of what was done.  What happens now is it may possibly save 3 files which are basically thumbnails of the front, left and right cameras.
[diagnostics.txt](https://github.com/marcone/teslausb/files/3570932/diagnostics.txt)

marcone, 2019-09-03:
Your diagnostics show there were no files to archive. The current code doesn't send notifications when there's nothing to archive.

curiousgrge, 2019-09-03:
Not for the last archive but there were certainly other events that had files to archive.  I have since taken the device out just so I can have it record sentry mode events until I work out the issues.  It appears that anytime it tries to save anything now would only be the 3 files that appear to be thumbnails only starting August 30 @ 8:55PM.

Isn't the low space message I'm getting every time now worrysome?

marcone, 2019-09-03:
> Isn't the low space message I'm getting every time now worrysome?

Not necessarily. It just indicates it can't extend your "RecentClips" recordings past 1 hour.
I'd recommend reinstalling from scratch, and using a much smaller size (like 16G or 24G) for your recordings, and also make your music drive as small as possible/needed, leaving the rest as free space.

DonC52, 2019-09-03:
Jumping in here...   
I am getting the same sort of errors as in the attached file...

[small.log](https://github.com/marcone/teslausb/files/3572014/small.log)

Are you saying that you can specify a size in the camsize parameter rather than a percent? (Like the musicsize parameter?)  I have a 64G SanDisk and currently about 1G or so of music (as shown on Windows - not sure how to tell on the Pi) and want to leave enough room on the saved clips area for a good week away from home.  How is the extra room not specified on these parameters used? You format it into another partition or disk image (i.e. backingfiles) or what?  You have any ideas on what to set the two parameters in my case? 

Any thoughts about expanding the Saved/Sentry area as well?  It would be cool to have the system expand into free space for saved/sentry clips so as not to be limited because your away from home. kind of dynamically allow the saved/sentry clips expand and the extra RecentClips collapse back to the 1 hour.

marcone, 2019-09-04:
> Are you saying that you can specify a size in the camsize parameter rather than a percent? (Like the musicsize parameter?)

musicsize and camsize are specified in exactly the same way: either as a percentage, or as a fixed size, e.g. "16G"

> I have a 64G SanDisk and currently about 1G or so of music (as shown on Windows - not sure how to tell on the Pi) and want to leave enough room on the saved clips area for a good week away from home.

That's going to be tough with  a card that small, I think. You're going to have less than 60 GB left for recordings, and the car records about 5.5 GB per hour. Let's call that 1 GB per Sentry event. So unless you expect to have fewer than 60 Sentry events during your week away from home, you should probably get a (much) bigger card.

> How is the extra room not specified on these parameters used? You format it into another partition or disk image (i.e. backingfiles) or what?

It's used to create reflinked (copy-on-write) snapshots of /backingfiles/cam_disk.bin. This allows preserving the current recordings while doing very little I/O and using virtually no extra space until the car starts overwriting the old recordings.

> You have any ideas on what to set the two parameters in my case?

It depends on whether you want to prioritize Saved/Sentry events or RecentClips. Currently I'm prioritizing RecentClips storage, since I come home every day. If you don't care that much about RecentClips but want to preserve all the SavedClips from Sentry events, then you should set the camsize to be as big as possible. You won't get the benefit of the extended RecentClips storage, but you'll give the car lots of space to store Sentry clips.

> Any thoughts about expanding the Saved/Sentry area as well? It would be cool to have the system expand into free space for saved/sentry clips so as not to be limited because your away from home. [...]

That won't work with the current mechanism, and I don't intend to change that. If you have a need to save many Sentry events, that's going to have to come at the expense of RecentClips, and that's something you'll have to decide up front.

DonC52, 2019-09-04:
When you say a much bigger card, what size are you thinking about? Like 128G, 256G?

You got a suggestion on make (SanDisk, Lexar, ...) and size?

DonC52, 2019-09-04:
Does the camsize parameter allocates for both the Recent and Security videos - correct?
So lets say I have 57G left over after a 1G music area. (A nice number!) Of that, 5.5G is gone for each hour of car recordings. That leaves about 51.5G (about 51 Sentry recordings.  So if I set the camsize less than 51G (lets say 47G) that will leave room for about 41 Sentry recordings and 10G of space for extra Recent recordings. About 1 extra hour of Recent recordings?

Do I got that all right?

curiousgrge, 2019-09-04:
Yeah I have dedicated 64G now for CAM.  Isn't the idea of archiving is to move the files to keep the CAM partition pretty much empty each time it archives?  It would take quite a bit of time before one would fill up that space if they come back home every night.  So for it to say that it is low on space tells me there is something wrong considering I never got that message prior to the latest build.  Perhaps you never considered that messaging in previous builds.

Part of the reason for not wanting to start from scratch is my somewhat large MUSIC partition which is almost 30G now.  This means having to sync the music again which takes a while and you lose favorites in the process.  If there is a way to at least retain favorites, I'm more inclined to start from scratch.

marcone, 2019-09-04:
> Yeah I have dedicated 64G now for CAM. Isn't the idea of archiving is to move the files to keep the CAM partition pretty much empty each time it archives? It would take quite a bit of time before [...]

The "low on space" message is new, and has nothing to do with the car side of things. If you set camsize to 64G, and assuming your sd card is big enough to accommodate that, then the car sees a 64G drive to write its recordings to.
The "low on space" message simply indicates that there is less than "camsize" diskspace left on the actual sd card, so in your case you would get that message when there's less than 64G of free space left (on the sd card /backingfiles partition, not on the drive that the car sees).

> Part of the reason for not wanting to start from scratch is my somewhat large MUSIC partition which is almost 30G now. This means having to sync the music again which takes a while and you lose [...]

I don't know where the car stores the favorites info (not on the drive itself, it appears), so I don't know how to preserve that info. You could possibly do something like renaming the music_disk.bin file to something else so that the setup script doesn't delete it, specify zero music size in the conf file, go through setup again, then rename the music file back.

curiousgrge, 2019-09-04:
I've already started the process of starting from scratch again and currently is syncing music files.  After that I will upgrade from the base 090615 build.

Allocating 24G for CAM and 32G for MUSIC.  Consider this closed at this point and will open another ticket if new issues arise.

closed 2019-09-04 by curiousgrge

marcone, 2019-09-04:
> Does the camsize parameter allocates for both the Recent and Security videos - correct?
> So lets say I have 57G left over after a 1G music area. (A nice number!) Of that, 5.5G is gone for each hour [...]

Not exactly. Due to the way the snapshotting mechanism works, you would get *up to* 1 extra hour of recent recordings, if you look at them at just the right time. But whenever teslausb goes to take another snapshot, it would first delete the previous snapshot, thus deleting those older recordings.

I don't have recommendations for specific cards, but I've tried Sandisk Extreme 128, Sandisk Ultra 256 and Samsung EVO Select 128.