marcone/teslausb #227: rclone.conf file being erased during ./setup-teslausb

DMBlakeley, 2019-09-04:
I went to create a new copy of TeslaUSB with the latest files and reconfiguring the CAM and MUSIC drives. Doing this I found out that the rclone.conf file nor the /mutable/configs/rclone directory existed after the setup process. 

With further investigation I found:
- rclone.conf file existed in the default location (/root/.config/rclone) after the rclone config process.
- the /mutable/configs/rclone directory creation, rclone.conf move, and link to this directory were being executed correctly during the ./setup-teslausb process.
- rclone.conf and associated directory did not exist after the ./setup-teslausb process completed.

Inspected the log during the ./setup-teslausb process I found that make-root-fs-readonly.sh ran after the initial rclone.conf move. This process mounted /mutable and recreated the configs directory. Although I could not find documentation if the "mount" command would erase previous contents I commented out lines 34 to 39 and reran the completed image and ./setup-teslausb procedure.

At the end of the ./setup-teslausb process I found that rclone.conf did exist in the /mutable/configs/rclone/ directory and "rclone config file" also provided the correct path.

I then issued a "halt" and unplugged the pi and reconnected.  After boot and mounting of the CAM folder I found that the /mutable/configs/ directory had been deleted! If I create the /mutable/configs/rclone directory and an rclone.conf file the archive process works correctly so all else seems to be correct.

Still looking but trying to figure out how the /mutable/configs directory is being deleted ... any thoughts?

DMBlakeley, 2019-09-04:
An additional piece of information. At present I can only create a teslausb configuration which supports rclone, ifttt, and Tesla keep awake with the DMBlakeley/teslausb fork which I pulled from macron/teslausb around August 17. I do need to create the /mutable/configs/rclone directory and the rclone.conf file after the TeslaUSB-setup process is complete.

Performing the same process with the macrone archive results in rclone not working.

At a loss at the moment to the root cause ...

marcone, 2019-09-04:
Not sure what's going on (I don't use rclone myself, and never tried it), but what you describe sounds like maybe the initial rclone config directory and file were created in the /mutable mountpoint directory before the mutable partition was actually mounted. Then once the partition is mounted, it will shadow the files in the mountpoint underneath.

DMBlakeley, 2019-09-05:
That seems like what is happening. Rclone is initially setup correctly in mutable, mutable is mounted and the files are not available.  Will try to figure out way to mount /mutable before the rclone.conf relocation.

Also not in the marcone archive that there was a recent change in the rclone/archive-clips.sh file to move the directory contents rather than each individual file. May explain some of my confusion.

Want to also add the kudos expressed by others on your efforts to continue to involve the functionality of teslausb. Hopefully v10 doesn’t cause too much headache!

marcone, 2019-09-05:
I think that maybe the following should be added to run/rclone_archive/verify-and-configure-archive.sh:
```
    if ! findmnt --mountpoint /mutable
    then
        mount /mutable
    fi
```
before it actually moves anything to /mutable

DMBlakeley, 2019-09-05:
That’s exactly the add I made to my fork of the run/rclone_archive/verify-and-configure-archive.sh file. Will check it out tomorrow.

marcone, 2019-09-05:
OK, I added that code to the script. I think that will fix the problem.

closed 2019-09-05 by marcone

DMBlakeley, 2019-09-05:
Code change fixed problem with rclone.conf being deleted. However, rclone/archive-clips.sh is not being triggered so files are not being transferred. Based on my fork the previous version of archive-clips.sh worked. Further investigation required.

DMBlakeley, 2019-09-06:
So here's the scoop ...

The last change to "rclone/archive-clips.sh" change the archive function from from moving just the movie files to events/files. Simulating on my Mac I found that I had to create the SavedClips folder with a dummy events folder such as "Dummy" and place some jpegs in this folder before testing with "touch /tmp/archive_is_unreachable". 

This is the same behavior as "cifs/archive-clips.sh" but not what I expected based on previous tinkering and also not explained as the reason for the "rclone/archive-clips.sh" change. May be transparent for the coding pros but not for us learning ...

DMBlakeley, 2019-09-06:
Made the latest marcone image with 16GB allocated on 64GB card last evening. Booted fine and saved a couple of events/clips touching the icon on the screen. When I got home the RPi connected to the network and transferred the files without issue. IFTTT messaging also worked fine. 

I want to take a look at the time syncing in archiveloop.sh (set_time) as this function takes quite a long time to run. Almost seems like the "Waiting for time to be set by ntpd..." section is not needed. Even after the time is set this loop fails and takes 30+ seconds to execute the second half to stop ntp, set time, start ntp executes to reset time.

marcone, 2019-09-07:
That initial section with the loop can probably be removed. It should be sufficient to stop the service, force the update, and restart the service.