marcone/teslausb #228: "Optional hotspot to access recordings while on the go"

aberen001, 2019-09-04:
I have this set up correctly and can access the pi as an AP. How would I go about viewing these recordings on the go?

AlbyZ, 2019-09-04:
[FileExplorer](https://apps.apple.com/au/app/fe-file-explorer-file-manager/id510282524)

The free version works.  You can only access one share, but thats good enough.

It has a built in mp4 media player, but its not a [TeslaCamPlayer](https://sentrycam.appspot.com/)

Problem is that the iPhone auto joins the WiFi, unless you 'forget'/remove the AP each time, during which your phone has no internet access.

marcone, 2019-09-04:
@aberen001 you'll need to have also configured SMB access to the Pi, but you can use a phone, tablet or laptop.

@AlbyZ Yes auto-join is super annoying. I thought iPhones have an option to disable auto-join though. Unfortunately stock Android doesn't, but some Samsung devices do. So check your wifi settings and see if there's an option.

closed 2019-09-04 by marcone

aberen001, 2019-09-04:
@marcone There is on the iPhone. If you click on the WiFi name in settings once you're connected, you can turn off the "Auto Join" setting.
I'll enable SAMBA and give that a shot, thank you!!