marcone/teslausb #232: Tesla API won't keep car awake

bladec59, 2019-09-05:
Have 2 cars (2018 MX and 2019 MS).  For neither car will the API keep the USB port powered.  It appears this may be a Tesla change; when I use the mobile app to start the air conditioning the same thing happens...no power to the USB ports.  I've tested the API manually using the tesla_api.py function and that seems to work.....get the expected responses.  Cars running 2019.28.3.1. or higher.

Not sure what options exist to resolve.

bladec59, 2019-09-05:
Possible solution:  use the set_sentry_mode command in the API to keep USB powered on;  after archiving is complete will then need to use same command to turn sentry mode off.

bladec59, 2019-09-06:
Was able to update the tesla_api.py script to add the set_sentry_mode command and then update archiveloop to execute 1 wake_up command followed by 1 set_sentry_mode command prior to running the archive_clips.sh script.  Then after the script I run a 2nd set_sentry_mode command to turn it off.  Seems to work correctly, and using sentry mode removes the need for a wake_up command every 2 minutes.  I'm just a 'duffer' of a programmer and luckily got this to work.  Happy to share actual changes if you want....am sure they don't meet good coding standards.

bdjdev, 2019-09-06:
I've been working on a patch which temporarily enables Sentry Mode to power the archive process.  I'm still testing the fix on my MX but will submit a PR for review.

marcone, 2019-09-07:
Thanks for the fix!

closed 2019-09-07 by marcone

bladec59, 2019-09-07:
Will self-update grab/install the updated files?

Closed #232<https://github.com/marcone/teslausb/issues/232>.

marcone, 2019-09-07:
> Will self-update grab/install the updated files?

`/root/bin/setup-teslausb selfupdate` will only update that script. You then need to run the setup-teslausb script to update everything else.

simonrb2000, 2019-10-13:
Is this fix in the latest scripts? I just setup a fresh stable install and currently only issue is the car isn't staying on and kills power to the USB port soon after I exit when I get home halting the archiving. The API details entered into the .config file are correct. 
Cheers