marcone/teslausb #234: stage7/Cmdline.txt fix

CR1AT0RS, 2019-09-05:
Issue:
Building image via Pi-gen fails due to mismatch using diff. 

>  Processing triggers for libc-bin (2.28-10+rpi1) ...
[15:51:02] End /home/meal/Downloads/pi-gen-master/stage7/00-teslausb-tweaks/00-packages
[15:51:02] Begin /home/meal/Downloads/pi-gen-master/stage7/00-teslausb-tweaks/00-patches
Converting meta-data to version 2
Applying patch /home/meal/Downloads/pi-gen-master/stage7/00-teslausb-tweaks/00-patches/01-cmdline-update.diff
patching file rootfs/boot/cmdline.txt
Hunk #1 FAILED at 1.
1 out of 1 hunk FAILED -- rejects in file rootfs/boot/cmdline.txt
Patch /home/meal/Downloads/pi-gen-master/stage7/00-teslausb-tweaks/00-patches/01-cmdline-update.diff does not apply (enforce with -f)

Fix: 
Update cmdline.txt to this:
`--- stage7/rootfs/boot/cmdline.txt
+++ stage7/rootfs/boot/cmdline.txt
@@ -1 +1 @@
-console=serial0,115200 console=tty1 root=ROOTDEV rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet init=/usr/lib/raspi-config/init_resize.sh
+dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=ROOTDEV rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet modules-load=dwc2,g_ether`

marcone, 2019-09-05:
I assume this builds a Buster-based image now, and that's the reason for the failure?

CR1AT0RS, 2019-09-06:
Here's the diff update for config.txt 

> --- stage7/rootfs/boot/config.txt
+++ stage7/rootfs/boot/config.txt
@@ -54,6 +54,7 @@
 
 # Enable audio (loads snd_bcm2835)
 dtparam=audio=on
+dtoverlay=dwc2
 
 [pi4]
 # Enable DRM VC4 V3D driver on top of the dispmanx display stack

kdvlr, 2019-09-06:
Was this the only change required to make Buster work ?

kdvlr, 2019-09-06:
I tried creating a Buster image, but it failed. Here's the log. Any suggestions appreciated

```
Now at patch /root/pi-gen/stage7/00-teslausb-tweaks/00-patches/04-hostname-update.diff
[16:28:28] End /root/pi-gen/stage7/00-teslausb-tweaks/00-patches
[16:28:28] Begin /root/pi-gen/stage7/00-teslausb-tweaks/00-run.sh
[16:28:28] End /root/pi-gen/stage7/00-teslausb-tweaks/00-run.sh
[16:28:28] End /root/pi-gen/stage7/00-teslausb-tweaks
[16:28:28] Begin /root/pi-gen/stage7/files
[16:28:28] End /root/pi-gen/stage7/files
[16:28:28] End /root/pi-gen/stage7
[16:28:29] Begin /root/pi-gen/export-image
[16:28:29] Begin /root/pi-gen/export-image/prerun.sh

Welcome to fdisk (util-linux 2.33.1).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Device does not contain a recognized partition table.
Created a new DOS disklabel with disk identifier 0x42f48939.

Command (m for help): Created a new DOS disklabel with disk identifier 0x8302d3a6.

Command (m for help): Partition type
   p   primary (0 primary, 0 extended, 4 free)
   e   extended (container for logical partitions)
Select (default p):
Using default response p.
Partition number (1-4, default 1): First sector (2048-4300799, default 2048): Last sector, +/-sectors or +/-size{K,M,G,T,P} (8192-4300799, default 4300799):
Created a new partition 1 of type 'Linux' and of size 256 MiB.

Command (m for help): Disk /root/pi-gen/work/2019-09-06-teslausb/export-image/2019-09-06-teslausb-lite.img: 2.1 GiB, 2202009600 bytes, 4300800 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x8302d3a6

Device                                                                           Boot Start    End Sectors  Size Id Type
/root/pi-gen/work/2019-09-06-teslausb/export-image/2019-09-06-teslausb-lite.img1       8192 532480  524289  256M 83 Linux

Command (m for help): Selected partition 1
Hex code (type L to list all codes): Changed type of partition 'Linux' to 'W95 FAT32 (LBA)'.

Command (m for help): Partition type
   p   primary (1 primary, 0 extended, 3 free)
   e   extended (container for logical partitions)
Select (default p):
Using default response p.
Partition number (2-4, default 2): First sector (2048-4300799, default 2048): Last sector, +/-sectors or +/-size{K,M,G,T,P} (540672-4300799, default 4300799):
Created a new partition 2 of type 'Linux' and of size 1.8 GiB.

Command (m for help):

Command (m for help):
Disk /root/pi-gen/work/2019-09-06-teslausb/export-image/2019-09-06-teslausb-lite.img: 2.1 GiB, 2202009600 bytes, 4300800 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x8302d3a6

Device                                                                           Boot  Start     End Sectors  Size Id Type
/root/pi-gen/work/2019-09-06-teslausb/export-image/2019-09-06-teslausb-lite.img1        8192  532480  524289  256M  c W95 FAT32 (LBA)
/root/pi-gen/work/2019-09-06-teslausb/export-image/2019-09-06-teslausb-lite.img2      540672 4300799 3760128  1.8G 83 Linux

Command (m for help): The partition table has been altered.
Syncing disks.

losetup: /root/pi-gen/work/2019-09-06-teslausb/export-image/2019-09-06-teslausb-lite.img: failed to set up loop device: Resource temporarily unavailable
```

CR1AT0RS, 2019-09-16:
Almost yes timedatectl update was also helpful in set_timezone function.

closed 2019-09-16 by CR1AT0RS