marcone/teslausb #240: Tracking for Raspberry Pi4 Support

kdvlr, 2019-09-06:
I have been getting the USB Too slow error far too frequently and have had issues with video recording and high bit rate music streaming on my PiZero earlier, so I decided to test out the Pi4 to see if I could use that. My plan was to use the SD card for boot and use a USB3.0  SSD that  I had to be the underlying partition.  Pi4 is the first full size raspberry pi to support USB OTG

I manually installed Buster, added dtoverlay=dwc2 to  config.txt  and was able to successfully mount a drive partitioned into  TESLACAM and MUSIC (did not use the scripts here). It seemed much faster (placebo alert !!). Primary test was by driving (to ensure teslacam was being recorded) and simultaneously shuffling very quickly through all my 8,000 song library. This for sure would have failed if I had done this on my Pi0. So far, with a half hour test, it looks good. 

Next step is to tryout the scripts from this repository and see what, if any, changes are required to make Pi4 work. I might just mount /backingfiles and /mutable to the SSD and see if the scripts take care of the rest. 

I was pleasantly surprised to see that the Pi4 was able to be powered directly from the Tesla USB port using a USB2.0 USB-A to USB-C cable (for the half hour that I tested). You cannot use a USB3.1 USB-C to USB-A cable  due to Raspberry PI's  well known USB-C implementation error. 

I am keeping this bug open so I can collaborate with others on  making this work.

bahree, 2019-09-06:
OK, I dont know enough of the cmdline.txt and how it impacts but been trying to get a pi4 to boot off a ssd and thats not working (separately see the [comments here](https://jamesachambers.com/raspberry-pi-4-usb-boot-config-guide-for-ssd-flash-drives/)). However when using the image itself, I can't get it to boot up at all - the pi4 doesn't recognize this.

@kdvlr did you boot / use a usb drive or to the sd card? And did you use steps outlined in the readme, with the exception of adding the dtoverlay? Sorry for the newbie questions.

kdvlr, 2019-09-06:
The default image will not work with Pi4 as it is based on Raspbian Stretch  and the Pi4 requires Buster. 

Pi4 does not support USB boot, so you will have to have /boot on SD card and then have the rootfs as the USB Drive which is what I did.

kdvlr, 2019-09-07:
Somehow the teslausb-setup is  failing  to  proceed  after  reading the variables. It gets stuck after reading config. 

```
root@teslausb:~/bin# bash -x ./setup-teslausb selfupdate
+ '[' ./setup-teslausb '!=' ./setup-teslausb ']'
+ '[' '' '!=' ./setup-teslausb ']'
+ FLOCKED=./setup-teslausb
+ flock -en -E 99 ./setup-teslausb ./setup-teslausb selfupdate
Read setup info from teslausb_setup_variables.conf (yes/no/cancel)? yes
reading config from /root/teslausb_setup_variables.conf
```

bahree, 2019-09-07:
I have the exact same issue but also on a pi zero. I thought it was related to my UTM but it's not. I do have a mesh network and don't know yet if it's that. Tried with two different devices and cards and same issue. See #211, I think it's related.

kdvlr, 2019-09-07:
In my case, i used ``` bash -x ./setup-teslausb``` that narrowed down the issue to curl failing due to SSL issue. It seems that happens occasionally on the Pi. I downloaded the image and flashed it.  Looking at #211 , it seems like the same issue. You can run the curl command manually without the -s and that should tell you why curl is failing. 

i have a fork /kdvlr/teslausb, branch Pi4 that I am using for this in case you're interested.

kdvlr, 2019-09-07:
Overall, it seems to work pretty good. I had to make the following changes

1. Remove the flashing leds part. That doesn't work.  
2. The create-backingfiles-partition.sh script needs to be changed. I manually created the partition. 
3. Any other references to /dev/mmc* should be changed.  
4. Somehow the dtoverlay=dwc2 line was not added to /boot/config.txt . I added it manually, but need to figure why it was not added. 
5.  After setup the ntp was not set correctly. Deleting /etc/systemd/system/multi-user.target.wants/ntp.service  which prevents ntp service from starting solved that.

joverdijk, 2019-09-08:
I would really like to use RPI4 for the teslausb/cam/music (using samsungT5 ssd via external usb), but beside that also do some scripting/apache/wifi-hotspot (teslaviewer etc).
I am having troubles mounting the tesla-cam image from backingfiles as readonly on the RPI to grab still-frames using ffmpeg out of the mp4 files that are constantly written.
I want to include live webcam into my domotica.
What is a good and stable way? I now just have a second mount in fstab for de image.bin as readonly but I think it is better to do it via a snapshot and loop-device / devmapper thingie?
But I'm not familiar with that. Someone ideass?

kdvlr, 2019-09-13:
@marcone - I think I am ready to push some changes to enable Pi4 support. Here are the changes. Let me know if you want me to include the checks in the existing files or move them to separate files. 

1. verify-configuration.sh - This can be left as-is for SD-card only setup (not tested by me). If using external disk, this script would have to be substantially changed. Thinking of adding a new flag saying USE_EXTERNAL_DISK='dev/sda' which specifies the external disk and a separate verify-configuration-external.sh that will verify space on the specified device. 

2. create-backing-files.sh - Similar to above, will need a different one when using external ssd. I can create a separate file for that. 

Those are the major changes I see. Let me know how you want to handle the release for this. I did not do the one-step setup. It will have to be a manual setup.

marcone, 2019-09-13:
So there are no changes needed to support Pi4, just to support an external drive?

kdvlr, 2019-09-14:
I believe so. I did not try the headless setup, so I cannot confirm for sure. The Pi4 also needs buster to boot up, so with the existing image, you will have to use manual setup for pi4. 

Otherwise, there is nothing OS specific that is being done. Pi4 is generally consistent with the same naming conventions.

NTP has been updated a bit, so the boot process takes a little longer.  

I deleted /etc/systemd/system/multi-user.target.wants/ntp.service to stop the ntp service from starting and used your script to set the time. I don't know if that's the right thing, but it is working. 

Everything else is  to support an external disk.

kdvlr, 2019-09-14:
Just verified the Wifi AP support as well and that works too.

marcone, 2019-09-22:
I uploaded a buster-based image at https://github.com/marcone/teslausb/releases. Only tested it with Pi0 though, would be nice if someone could try if works on Pi4, and if not, what is needed to make it work.

GCustom, 2019-09-24:
the operation on the car side is much better than it was on the pi zero w but I still have a bunch of 0kb files uploaded. This might be on my NAS side of things, I'll check some things out because it looks like it is actually ok.

Does the uploader delete right away or does it track which files/folders have been uploaded and delete them when space is needed?

marcone, 2019-09-24:
The archiver deletes the files as soon as they've been archived.

GCustom, 2019-09-24:
I got several days uploaded with 0kb files for all of them. 

The Pi Zero W is near the PC and I just verified that yes, everything except the files saved by FSCK and the one hour loop are deleted. I don't think there is any way to detect that this is happening unless the fileshare is throwing errors at the Pi during the copy. So a word of warning for other users is keep an eye on your target directory when you get a space warning.

kdvlr, 2019-09-24:
Are you in a hot area? I noticed a pattern with my vehicle. I left it out in a 100 degree day outside and I got 0kb files on the camera that was  facing the sun. I repeated the test and the same thing happened.

kdvlr, 2019-09-24:
The buster image did not work for me. It  created the enable_wifi.sh script and stopped. Rerunning /etc/rc.local did not help either. 

```
 Sun 22 Sep 22:54:53 BST 2019 : Detecting whether to update wpa_supplicant.conf
Sun 22 Sep 22:54:53 BST 2019 : Wifi variables specified, and no /boot/WIFI_ENABLED. Building wpa_supplicant.conf.
Sun 22 Sep 22:54:53 BST 2019 : Rebooting...
Sun 22 Sep 22:55:10 BST 2019 : Detecting whether to update wpa_supplicant.conf
Tue 24 Sep 19:21:28 BST 2019 : Detecting whether to update wpa_supplicant.conf
```

marcone, 2019-09-24:
That makes no sense to me. That message is printed by rc.local, but only if it did not enter [this](https://github.com/marcone/teslausb/blob/main-dev/pi-gen-sources/00-teslausb-tweaks/files/rc.local#L101) if-block, however it should have entered that because /boot/TESLAUSB_SETUP_FINISHED did not exist and you set HEADLESS_SETUP to "true".

kdvlr, 2019-09-24:
At a quick glance. I did not understand too. Thats why I made sure to include the steps to show that there was no TESLAUSB_SETUP_FINISHED file as well as the env variable set. Will look into this further.

GCustom, 2019-09-24:
Yes it was hot but it was 0 kb for all and the Pi was empty except for the recent directory and the fsck files. Will bring the Pi4 in to see if it is showing anything different than the Pi 0W did

marcone, 2019-09-24:
I seem to remember a previous conversation about 0-byte files being a car issue, resolved by rebooting the MCU.

GCustom, 2019-09-24:
Rebooted MCU and powered down car for a while just now but on a straight USB drive it did record normally

kdvlr, 2019-09-27:
Submitted PR https://github.com/marcone/teslausb/pull/275 for this. Additional testing would be much appreciated.

GCustom, 2019-09-27:
The Pi4 had all the files but the Pi 0 was mostly empty

elight, 2019-10-01:
Has there been any testing of the one step setup?

kdvlr, 2019-10-02:
I tested the one step, but so far seems n=1. Do try it out and join us on discord if you have errors.

kdvlr, 2019-10-02:
> The Pi4 had all the files but the Pi 0 was mostly empty

Could you  clarify this please?

kdvlr, 2019-10-06:
Closing this as support has been added.

closed 2019-10-06 by kdvlr