marcone/teslausb #248: Ongoing issue keeping power on USB ports

bladec59, 2019-09-09:
Upgraded to latest release to incorporate the set_sentry_mode functionality to keep the car awake so all files can be archived.  However, there appears to be something still not quite right.  Attached is a section of the archiveloop.log today.  After about 2 minutes the archiving stops, likely due to the power shutting off.  Also, the logs don't have any reference to the API being called to enable Sentry Mode.  

In further testing with a small snippet of code I find that including the /root/bin/send-push-message for some reason causes the rest of the test code to be skipped....and there is a push command right before the code to set_sentry_mode.  Perhaps I'm completely off-base, but there seems to be a connection there.

Will continue to investigate to determine root cause.

[TeslaUSB-issue with API.txt](https://github.com/marcone/teslausb/files/3592713/TeslaUSB-issue.with.API.txt)

bladec59, 2019-09-10:
did a complete re-flash and reinstall.  Working at the moment...will keep an eye on it.

bladec59, 2019-09-10:
Now I see that an update was made to address the capitalization of the sentry mode status results...perhaps that was my problem.  Will close this issue soon if everything stays ok.

bdjdev, 2019-09-11:
> Now I see that an update was made to address the capitalization of the sentry mode status results...perhaps that was my problem. Will close this issue soon if everything stays ok.

Yep, sorry about that... last minute change on my part.  Let me know if this is still not working for you.

bladec59, 2019-09-11:
Seems to be ok now; closing issue.

closed 2019-09-11 by bladec59