marcone/teslausb #249: Passwords

DonC52, 2019-09-10:
Can the passwords on the Pi be changed without issue?  (root, pi, ...)  Would add to the security of the system if they could be changed from known to user specified - kind of important for access enabling the WiFi access point option. I could update the README with some additional verbiage about changing the mounts to read/write and changing the passwords with passwd.

mickymix, 2019-09-10:
If I'm not wrong (and I did it personally two days ago), it is as simple as:
```
sudo -i
/root/bin/remountfs_rw
passwd
```

DonC52, 2019-09-10:
I know it is very simple to do, but I wanted to be sure that nothing in all of the scripts require the passwords to be what they are installed as.  I don't want it to error out when doing something not normally done. ;-)

marcone, 2019-09-10:
The scripts don't require the default password, it's perfectly fine to change it.
In fact they don't even care there is a "pi" user at all, so you can add a new user (don't forget to add it to the "sudo" group), and then delete the "pi" user.

DonC52, 2019-09-11:
I added a section to README and submitted a merge request.

marcone, 2019-09-12:
Merged

closed 2019-09-12 by marcone