marcone/teslausb #250: Features request: further WiFis and Self-Update settings

mickymix, 2019-09-10:
It took me a lot of reading to understand that it is not necessary to reinstall everything once a new version or fix is available, but some commands can be run to have this performed (still have to note down correctly which ones, but I know they are there...).
And while I have my NAS at home saving my clips, I would like to perform some other maintenance tasks while in the office, so a second WiFi connection would be nice.
But I'm not very comfortable on PI's command line, so it's like walking on a mined path for me...
I would appreciate if a second (or even third) WiFi connection could be setup, possibly using the teslausb_setup_variables.conf (for initial users) or with a further script to be run on request (for those who already have a working setup).
And it would be nice (and definitely confirming such presence) a setting for requesting a selfupdate and/or a complete update after clips transfer, which can be run daily and/or weekly and/or never...
Sorry for my long text... :-)

marcone, 2019-09-10:
Thanks for the suggestions, but I think this adds too much complexity and risk for not much benefit. You'd have to take into account wifi connections being added or removed, and automatic update is risky because it's more likely that wifi will go away during the update and leave things in a broken state.

closed 2019-09-10 by marcone

mickymix, 2019-09-10:
Thanks for considering and taking time to answer.
I would only point out then that a brief resume of how upgrades can be manually performed would be nice (really I only discovered them after one month of reading around and trying...).
And a quick hint on how to manually add a second WiFi connection would be appreciated. ;-P

marcone, 2019-09-10:
There is a section in the README that describes how to make the root partition writable so you can make changes to the system after setup  (https://github.com/marcone/teslausb#optional-making-changes-to-the-system-after-setup), but 
it's beyond the scope of this project to provide guides on "how to do X on linux", so you'll have to look that up yourself. For wifi, search for "wpa_supplicant".

mickymix, 2019-09-10:
Thanks again, for all the hints...
Anyway it seems I'm quite "old style" as I was not referring to explain how to do things in Linux, but just to point out that even installing an old TeslaUsb release (as the latest headless setup build is), it performs downloads of the latest available version during setup, building an up-to-date system.
This was not so clear to me, but probably it's just because I'm too much "old style"...

marcone, 2019-09-10:
I've reorganized the wiki a bit and added an item about updating by running setup again.