marcone/teslausb #254: Add scan_ssid=1 to wpa_supplicant by default or option to enable it in config

dprus, 2019-09-13:
The WiFi SSID I use for my IOT devices is hidden. In order for wpa_supplicant to work with networks with hidden SSIDs you need to add the option "scan_ssid=1".

As a workaround to this issue I added "scan_ssid=1" into my wpa_supplicant.conf.sample file but it may not be obvious to anyone with a network with a hidden SSID that this is what's needed.

I'm not sure of what the impact is of having "scan_ssid=1" in the wpa_supplicant.conf by default for people who do not use networks with hidden SSIDs.

Anyway, it would be great if either:
1. There is an option in teslausb_setup_variables.conf to specify your network has a hidden SSID that would then add "scan_ssid=1" to wpa_supplicant or;
2. "scan_ssid=1"  was added to wpa_supplicant.conf by default or;
3. The WIKI documentation pointed out the workaround I used

marcone, 2019-09-13:
I don't want to add this by default, because according to https://w1.fi/cgit/hostap/plain/wpa_supplicant/wpa_supplicant.conf the scan_ssid option "will add latency to scanning, so enable this only when needed.'
PR welcome.

paulstaylor, 2019-09-16:
I work in network security.  I don’t recommend that anyone hide their SSID.  It doesn’t add any real security.  If someone wants to hack your network, they can find hidden SSIDs easily.  Here’s an article about it.  https://www.howtogeek.com/howto/28653/debunking-myths-is-hiding-your-wireless-ssid-really-more-secure/

dprus, 2019-09-16:
Honestly, I don't do it for security. I do it to remove noise. I have separate family,  guest and IOT networks and I just don't like all the networks showing up in a scan.

I'm aware that it's not secure. Other than, perhaps,  go keep the honest honest :)

-Richard.

paulstaylor, 2019-09-16:
Fair enough...

txex95, 2019-09-18:
Maybe worth including as a headless setup option? Commented out by default.

marcone, 2019-11-02:
As noted above, "PR welcome", but since there's been no PR so far I figure this isn't something anyone feels strongly about. Since I don't plan to add this myself, I'm closing this issue.

closed 2019-11-02 by marcone