marcone/teslausb #256: "dummy" archiveloop ($ARCHIVE_SYSTEM=none) missing lots of important bits

fat-tire, 2019-09-14:
So if you choose to NOT sync your archive, configure.sh [creates a "dummy" archiveloop](https://github.com/marcone/teslausb/blob/5683c4d02cfd70b06170fd94d1eb4d8e8f3d176f/setup/pi/configure.sh#L354):

     	#!/bin/bash -eu
	modprobe g_mass_storage

Unfortunately, this is a pretty broken "loop".  It's missing a bunch of stuff:

* turning off /usr/bin/tvservice
* checking if the usb gadget is loaded
* ensuring the mounts were successful
* fixing any corruption/errors in the mountpoints
etc.

Practically speaking, stuff inevitably gets corrupt and isn't fixed, so the whole thing gets broken.

marcone, 2019-09-15:
I'll grant you that not configuring archiving isn't very useful, but I have no plans to make "no archiving" a supported mode of operation. Pull requests welcome though.
(it's not hard to add a no-op archiving method and associated scripts, but then the card would eventually fill up with saved clips)

closed 2019-09-15 by marcone

fat-tire, 2019-09-15:
But "none" is literally the [default](https://github.com/marcone/teslausb/search?q=none&unscoped_q=none)...

marcone, 2019-09-15:
Sure, and until recently that mode didn't do anything. Would you like me to change it back? :)

In any case, it looks like the new V10 Tesla firmware might solve the issue of the card filling up with saved clips, so I might revisit this when I get that update.