marcone/teslausb #258: Backingfiles problem

BrandX3, 2019-09-15:
I'm unable to get past this point below while doing a headless install.

The WIFI_ENABLED file exists, as does TESLAUSB_SETUP_STARTED.

~ ~
Fixing the modules-load parameter in /boot/cmdline.txt...
Fixed cmdline.txt.
using existing backingfiles and mutable partitions
TARGET        SOURCE         FSTYPE OPTIONS
/backingfiles /dev/mmcblk0p3 xfs    rw,noatime,attr2,inode64,noquota
Creating backing disk files.
create-backingfiles: starting
create-backingfiles: starting
create-backingfiles: cam: 20GB, music: 100%, mountpoint: /backingfiles
create-backingfiles: cam: 20GB, music: 100%, mountpoint: /backingfiles
archiveloop: no process found
umount: /mnt/cam: mountpoint not found
umount: /mnt/music: mountpoint not found
/tmp/create-backingfiles.sh: line 32: 20*1024*1024*1024B: value too great for base (error token is "1024B")
/tmp/create-backingfiles.sh: line 62: / 1024 : syntax error: operand expected (error token is "/ 1024 ")

[diagnostics.txt](https://github.com/marcone/teslausb/files/3613264/diagnostics.txt)

marcone, 2019-09-15:
Looks like you specified "GB" as the size multiplier, but that should be "G"

closed 2019-09-15 by marcone

BrandX3, 2019-09-15:
That fixed it. Thank you!