marcone/teslausb #264: Rsync trough Nas won't work

RSickenberg, 2019-09-17:
```
Tue 17 Sep 19:22:40 BST 2019: Archiving...
Tue 17 Sep 19:22:40 BST 2019: Disconnecting usb from host...
Tue 17 Sep 19:22:40 BST 2019: Disconnected usb from host.
Tue 17 Sep 19:22:40 BST 2019: Starting recording archiving...
Tue 17 Sep 19:22:40 BST 2019: Sending IFTTT message for moved files.
Tue 17 Sep 19:22:41 BST 2019: Ensuring cam file is mounted...
Tue 17 Sep 19:22:41 BST 2019: Mounting /mnt/cam...
Tue 17 Sep 19:22:41 BST 2019: Mounted /mnt/cam.
Tue 17 Sep 19:22:41 BST 2019: Ensured cam file is mounted.
Tue 17 Sep 19:22:41 BST 2019: Running fsck on /mnt/cam...
Tue 17 Sep 19:22:41 BST 2019: | fsck from util-linux 2.29.2
Tue 17 Sep 19:22:44 BST 2019: | fsck.fat 4.1 (2017-01-24)
Tue 17 Sep 19:22:44 BST 2019: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Tue 17 Sep 19:22:44 BST 2019: |  Automatically removing dirty bit.
Tue 17 Sep 19:22:44 BST 2019: | Performing changes.
Tue 17 Sep 19:22:44 BST 2019: | /dev/loop0: 1586 files, 1211099/1841695 clusters
Tue 17 Sep 19:22:44 BST 2019: Finished fsck on /mnt/cam.
Tue 17 Sep 19:22:44 BST 2019: wake up car process is 2318
Tue 17 Sep 19:22:44 BST 2019: Archiving through rsync...
Tue 17 Sep 19:22:45 BST 2019: No files to archive through rsync.
Tue 17 Sep 19:22:45 BST 2019: killing PID 2318
Tue 17 Sep 19:22:45 BST 2019: Unmounting /mnt/cam...
Tue 17 Sep 19:22:46 BST 2019: Unmounted /mnt/cam.
Tue 17 Sep 19:22:46 BST 2019: Finished archiving.
Tue 17 Sep 19:22:46 BST 2019: Music archive not configured or unreachable
Tue 17 Sep 19:22:46 BST 2019: Connecting usb to host...
Tue 17 Sep 19:22:46 BST 2019: Connected usb to host.
Tue 17 Sep 19:22:47 BST 2019: Waiting for archive to be unreachable...
```

I tried all the setup without passwords, so if I ssh trough my Pi zero W to my Nas it works on root.

Can somebody help me?

marcone, 2019-09-17:
```
Tue 17 Sep 19:22:45 BST 2019: No files to archive through rsync.
```
This means there is nothing for the Pi to archive. You'll need to have some Sentry or manually saved videos first.

closed 2019-09-17 by marcone

RSickenberg, 2019-09-17:
@marcone Thanks for the fast message, but in fact, there is plenty of sentry videos to save ...

After digging, it was an issue on my Synology, just had to `chmod 777` my shared folder.

marcone, 2019-09-17:
Ah you're right, that message is logged when nothing was archived, not when there's nothing to archive. Confusing. I should fix the messaging at least.