marcone/teslausb #265: Is there a way to change partition size and/or settings after setup?

yarzy1, 2019-09-17:
As the title says, is it possible? If so, can you point me to the documentation? Thanks!

marcone, 2019-09-17:
The answer is, as often is the case, "it depends".
Which "partitions" are you talking about? Which settings?

yarzy1, 2019-09-18:
The cam and music partition sizes after the setup have completed. So if I need more space for additional music, do I need to redo the entire thing? As far as settings, settings to keep the car awake. Can that be changed after setup has completed?

marcone, 2019-09-18:
The cam and music drives aren't actually partitions on the sd card, they are disk image files on the "backingfiles" partition of the sd card, which are then exported from the Pi as separate USB drives.
As for resizing them: you will have to delete the disk image files, then rerun setup-teslausb.

Most other settings can be changed just by updating the .conf file (which setup will have moved to /root) and then rerunning setup-teslausb.

closed 2019-09-18 by marcone