marcone/teslausb #267: WPA passwords with slashes in them make rc.local very sad

dopeghoti, 2019-09-19:
This may be rather complex to fix since you're using `sed` to write the `wpa_supplicant.conf` file, but the WPA password for my wireless network has `/`s in it, and that caused the first-boot to fail when running `rc.local`:

```
rc.local[464]: sed: -e expression #1, char 19: unknown option to `s'
[FAILED] Failed to start /etc/rc.local Compatibility
```

I was able to work around this by manually adding my password to the partially written WPA config file, booting, and manually `touch`ing `WIFI_ENABLED`, and then re-running `/etc/rc.local` seemed to kick things back off successfully.

marcone, 2019-09-19:
You can/should escape forward slashes with a backslash. This is mentioned in https://github.com/marcone/teslausb/blob/main-dev/doc/OneStepSetup.md

closed 2019-09-19 by marcone

dopeghoti, 2019-09-19:
I did try escaping them, as well as wrapping it in strong quotes before resorting to manual intervention.