marcone/teslausb #269: V10 - Errors with Pi Dashcam on Early Access build 2019.32.2.12

SemoTech, 2019-09-22:
Looks like the latest update V10 Early Access build 2019.32.2.12 that dropped for my TM3 last night has introduced some other changes to the DashCam/Sentry system.

While my setup has worked perfectly till now, I started to get "usb" errors.  The errors alternate between "[USB Drive is Not accessible by Dashcam](https://i.imgur.com/WCvIxYh.jpg)" and "[USB write speed is to low for Dashcam](https://i.imgur.com/SgM4pCc.jpg)". The notice specifies mandatory 4MB/s or better performance.

The SD card I am using in the Pi is the [128GB Sandisk Extreme U3](https://www.amazon.com/gp/product/B07FCMKK5X) which is both high endurance and high-speed. It is UHS Speed Class 3 (U3) and Video Speed Class 30 (V30) @ 160MB/s, so I doubt that the mSD card is the issue.  I suspect the Pi Zero W CPU being the bottleneck, especially after the AP and SAMBA shares features are enabled.  I wonder if disabling them would help?

Any way to benchmark, and also to improve the write performance?

XmemphistoX, 2019-09-22:
I got the same thing, I saw the errors more frequently when I had AP/Samba enabled.  However I still see the error from time to time with the features off.

I also didn't have these errors from a month ago.  Just after I updated to the newest teslausb that had the AP/Samba features included.

marcone, 2019-09-22:
This doesn't really seem any different from previous reports such as https://github.com/marcone/teslausb/issues/207

japhule, 2019-09-25:
I don't think the Sandisk Extreme cards (despite the name) are considered high endurance like the Samsung Pro Endurance cards. Extreme just means that it can write fast for HD video recordings and withstand some extreme environments, but it doesn't mean that it's ideal for dashcam/security cam constant recordings. It doesn't say endurance anywhere in its description either. No issues here with low speed errors using the Samsung Pro Endurance with the Pi and would recommend those cards.

marcone, 2019-09-25:
I've been using Sandisk Extreme for months without issue, and really at the rate that the car is writing to it (1.5 MB/s in v9, probably around 2 MB/s in v10 which I'm told also records the rear camera) it should not be a problem.
The "drive too slow" problem is probably some combination of the USB 2.0 interface, the mass storage emulation overhead, and the fact that the Pi just isn't that fast at writing to the sd card to begin with (it gets nowhere near the max card speed)

SemoTech, 2019-09-25:
@marcone is 100% correct.  Besides the Sandisk Extreme is a U3 card while the Samsung Pro Endurance is U1.  If you had to get a U3 endurance card the [Sandisk High Endurance](https://www.amazon.com/dp/B07NY23WBG) is the best of both worlds. 
But as @marcone pointed out, this is likely the USB 2.0 and overhead on the Pi.  I wonder if there are any optimizations we can do to achieve at least 5-6MB/s sustained and eliminate the warning.

For reference this AM I got the "USB drive not accessible by Dashcam" again and when I checked at home last night there have been no videos copied for the past several days following the new Tesla Firmware update.  That may be due to the folder changes Tesla implemented as the only events were Sentry.

@marcone does the latest code account for the new Sentry video folder changes?

marcone, 2019-09-25:
No, I have not updated the code yet because I don't have v10 yet.

As for optimizing to achieve higher throughput, I don't know what could be done there. It's not teslausb that provides the drive emulation, it's linux itself, via the g_mass_storage module. Other than "nofua", g_mass_storage doesn't appear to have any tuneable parameters that would affect performance.

japhule, 2019-09-25:
I don't get the drive too slow warning using the Samsung Pro Endurance. Would be more than happy to test a new version that gets the sentry folder.

SemoTech, 2019-09-25:
Thanks @marcone I did not think there was a fix for the g_mass_storage module, but was hoping that maybe Buster made some improvements. I will try it as soon as we get the folder structure fixed.

@japhule & @marcone I will swap the Pi for a spare/backup Samsung USB, and record both Dashcam and Sentry events, to get the folder structure and report it here.

normalfaults, 2019-09-25:
Note additional rear_view camera video ...

`TeslaCam/RecentClips/yyyy-mm-dd-hh-mm-ss-<front/left_repeater/right_repeater/rear_view>.mp4`
`TeslaCam/SavedClips/yyyy-mm-dd_hh-mm-ss/yyyy-mm-dd-hh-mm-ss-<front/left_repeater/right_repeater/rear_view>.mp4`
`TeslaCam/SentryClips/yyyy-mm-dd_hh-mm-ss/yyyy-mm-dd-hh-mm-ss-<front/left_repeater/right_repeater/rear_view>.mp4`

SemoTech, 2019-09-25:
> Note additional rear_view camera video ...
>
> `TeslaCam/RecentClips/yyyy-mm-dd_hh-mm-ss-<front/left_repeater/right_repeater/rear_view>.mp4` [...]

Thanks @normalfaults! So we have 3 folders now "**RecentClips**", "**SavedClips**" & "**SentryClips**" and the rear camera is recording as well.  Saved me from buying one :-). I also bet the rear camera video quality is better than the others as it was designed for human not AI use. 

@marcone Do you need anything else for a V10 compatible teslausb code update?

marcone, 2019-09-26:
> @marcone Do you need anything else for a V10 compatible teslausb code update?

Well it would be nice to have v10 so I can actually test the code 😄 
I put some untested changes in the "v10" branch. Feel free to give it a try.

SemoTech, 2019-09-26:
Thank you @marcone I will re-flash and try tonight.

For reference, swapping to a regular USB 3.0 Samsung 64GB USB stick has NOT yielded the warnings from Tesla.

SemoTech, 2019-09-26:
Never mind @marcone I think I got it.  Looks like we need to change:

`# export BRANCH=main-dev`
to
`export BRANCH=v10`

in the "teslausb_setup_variables.conf" file.

SemoTech, 2019-09-26:
@marcone Ran into small snag with buster after writing latest (buster-20190922) image to microSD card and updating the config file to BRANCH=v10. 
All seems to be processed fine but the "CAM" share never appeared on my Mac.  

teslausb-headless-setup.log seemed to show all was OK:

> ...
Thu 26 Sep 19:17:37 EDT 2019 : verify-and-configure-archive: Configuring the archive...
Thu 26 Sep 19:17:37 EDT 2019 : verify-and-configure-archive: Configured the archive.
Thu 26 Sep 19:17:37 EDT 2019 : configure: Skipping rc.local installation
Thu 26 Sep 19:17:37 EDT 2019 : make-root-fs-readonly: start
Thu 26 Sep 19:17:37 EDT 2019 : make-root-fs-readonly: Removing unwanted packages...
Thu 26 Sep 19:17:49 EDT 2019 : make-root-fs-readonly: Installing ntp and busybox-syslogd...
Thu 26 Sep 19:17:54 EDT 2019 : make-root-fs-readonly: Configuring system...
Thu 26 Sep 19:17:55 EDT 2019 : make-root-fs-readonly: done
Thu 26 Sep 19:17:55 EDT 2019 : Upgrading installed packages...
Thu 26 Sep 19:18:03 EDT 2019 : All done.

I rebooted and then manually ran /root/bin/setup-teslausb and got some warnings about name resolution as follows:

> root@ELECTRA-TM3USB:~# /root/bin/setup-teslausb
Read setup info from teslausb_setup_variables.conf (yes/no/cancel)? yes
reading config from /root/teslausb_setup_variables.conf
Configuring the hostname...
Configured hostname: ELECTRA-TM3USB
/root/bin/setup-teslausb is up to date
Updating package index files...
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Reading package lists... Done
sudo: unable to resolve host ELECTRA-TM3USB: Temporary failure in name resolution
sudo: unable to resolve host ELECTRA-TM3USB: Temporary failure in name resolution
sudo: unable to resolve host ELECTRA-TM3USB: Name or service not known
sudo: unable to resolve host ELECTRA-TM3USB: Name or service not known
sudo: unable to resolve host ELECTRA-TM3USB: Name or service not known
sudo: unable to resolve host ELECTRA-TM3USB: Name or service not known

Subsequently attempting to get diagnostics failed:

> root@ELECTRA-TM3USB:/boot# /root/bin/setup-teslausb diagnose > /mutable/diagnostics.txt
Read setup info from teslausb_setup_variables.conf (yes/no/cancel)? yes
cat: /tmp/root3405/var/log/boot.log: No such file or directory

Any suggestions to get the "CAM" share and possibly solve the hostname errors?

marcone, 2019-09-27:
Buster doesn't have /var/log/boot.log anymore, so the diagnostics code has to be updated. No idea what that "unable to resolve host" is about though. What does /mutable/archiveloop.log say?

SemoTech, 2019-09-27:
So strange, _this time it worked BOTH on the Mac and in the car!_ 

**Only thing still not working is the AP mode**, even tho the config file has been modified to enable the SSID and the password.  The SAMBA was NOT enabled as I figured I just need "remote SSH"

Fore reference here is the output of /mutable/archiveloop.log

> Thu 14 Feb 05:12:48 EST 2019: Starting archiveloop...
Thu 14 Feb 05:12:49 EST 2019: mounted all snapshots
Thu 14 Feb 05:12:49 EST 2019: low space, deleting /backingfiles/snapshots/snap-000000
Thu 14 Feb 05:12:55 EST 2019: warning: low space for snapshots
Thu 14 Feb 05:12:55 EST 2019: taking snapshot of cam disk: /backingfiles/snapshots/snap-000000/snap.bin
Thu 14 Feb 05:13:00 EST 2019: took snapshot
Thu 14 Feb 05:13:05 EST 2019: comparing /backingfiles/snapshots/snap--00001/snap.bin.toc and /backingfiles/snapshots/newsnap/snap.bin.toc
Thu 14 Feb 05:13:05 EST 2019: making links for /backingfiles/snapshots/newsnap/mnt, retargeted to /backingfiles/snapshots/snap-000000/mnt
Thu 14 Feb 05:13:05 EST 2019: made all links for /backingfiles/snapshots/newsnap/mnt
Thu 14 Feb 05:13:05 EST 2019: Waiting for time to be set by ntpd...
Thu 14 Feb 05:13:32 EST 2019: Time still not set, attempting to force it
Thu 26 Sep 20:16:02 EDT 2019: Archiving...
Thu 26 Sep 20:16:02 EDT 2019: Disconnecting usb from host...
Thu 26 Sep 20:16:03 EDT 2019: Disconnected usb from host.
Thu 26 Sep 20:16:03 EDT 2019: Ensuring cam archive is mounted...
Thu 26 Sep 20:16:03 EDT 2019: /mnt/archive is already mounted.
Thu 26 Sep 20:16:03 EDT 2019: Ensured cam archive is mounted.
Thu 26 Sep 20:16:03 EDT 2019: Checking saved folder count...
Thu 26 Sep 20:16:03 EDT 2019: Ensuring cam file is mounted...
Thu 26 Sep 20:16:03 EDT 2019: Mounting /mnt/cam...
Thu 26 Sep 20:16:03 EDT 2019: Mounted /mnt/cam.
Thu 26 Sep 20:16:04 EDT 2019: Ensured cam file is mounted.
Thu 26 Sep 20:16:04 EDT 2019: Running fsck on /mnt/cam...
Thu 26 Sep 20:16:04 EDT 2019: | fsck from util-linux 2.33.1
Thu 26 Sep 20:16:10 EDT 2019: | fsck.fat 4.1 (2017-01-24)
Thu 26 Sep 20:16:10 EDT 2019: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Thu 26 Sep 20:16:10 EDT 2019: |  Automatically removing dirty bit.
Thu 26 Sep 20:16:10 EDT 2019: | Performing changes.
Thu 26 Sep 20:16:10 EDT 2019: | /dev/loop1: 2 files, 2/3604191 clusters
Thu 26 Sep 20:16:10 EDT 2019: Finished fsck on /mnt/cam.
Thu 26 Sep 20:16:11 EDT 2019: There are 0 event folder(s) with 0 file(s) to move.
Thu 26 Sep 20:16:11 EDT 2019: Unmounting /mnt/cam...
Thu 26 Sep 20:16:11 EDT 2019: Unmounted /mnt/cam.
Thu 26 Sep 20:16:11 EDT 2019: Finished archiving.
Thu 26 Sep 20:16:11 EDT 2019: Music archive not configured or unreachable
Thu 26 Sep 20:16:11 EDT 2019: unmounting /mnt/archive
Thu 26 Sep 20:16:12 EDT 2019: Connecting usb to host...
Thu 26 Sep 20:16:12 EDT 2019: Connected usb to host.
Thu 26 Sep 20:16:12 EDT 2019: Waiting for archive to be unreachable...

marcone, 2019-09-27:
Seeing the same thing wrt to AP not working on Pi4. No idea why yet.

SemoTech, 2019-09-27:
Exactly @marcone no AP functionality on Buster. 

The Pushover Notifications are Awesome tho and looks like archiving works nicely. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

SemoTech, 2019-09-27:
All problems solved! 

AP functionality failure was due to blank spaces when uncommenting the lines for SSID and Password. 

Hostname issues have all been solved by adding:

127.0.1.1   Teslausb

to the /etc/hostname file. 

v10 code confirmed working @marcone you may want to roll it into main-dev.

normalfaults, 2019-09-27:
Is this on the buster image ?

SemoTech, 2019-09-28:
Yes @normalfaults v10 branch.

closed 2019-09-28 by SemoTech

mattster98, 2019-09-30:
Started getting the "USB is too slow 4MB/sec" warning.  Is there a list of known good SD cards forming, or is it still a limitation in the emulation?

firedfly, 2019-09-30:
@mattster98  - I've been using a 128gb Samsung Pro Endurance.  I received that message once with a early access release.  Haven't received that message with the public build.

mattster98, 2019-10-02:
Upgraded to the card you recommended (but 64GB version) and it's been working fine, until this morning.  Had the "slow USB" error when I got in the car.  A quick reboot of the TeslaUSB returned it to normal.

miles267, 2019-10-03:
> All problems solved!
> 
> AP functionality failure was due to blank spaces when uncommenting the lines for SSID and Password.
> 
> Hostname issues have all been solved by adding:
> 
> 127.0.1.1 Teslausb
> 
> to the /etc/hosts file.
> 
> v10 code confirmed working on Buster image @marcone you may want to roll it into main-dev.

For a new install with v10, should I be using the buster image with branch=main-dev (the default) or setting it to branch=v10?

marcone, 2019-10-03:
I deleted the v10 branch yesterday

miles267, 2019-10-03:
> I deleted the v10 branch yesterday

OK.  I am trying to get the Dashcam icon to appear on v10 (2019.32.11.1). Installed the buster image with branch=main-dev.  All connectivity works, exception dashcam icon never appears.  It's a fresh installed.  Even tried it multiple times.  Was hoping the branch=v10 might fix the issue as it appears in this thread it has for others.

SemoTech, 2019-10-03:
@miles267 It takes a long time on a Pi0W to process, update and be ready the 1st time. Have you run through the image process, copied the personalized config file to /boot and then booted the Pi but waited till the computer sees the “CAM” share? If so, just do a “shutdown -h now” and then plug it in the Tesla to boot. It should work just fine with latest V10 firmware, I have the same one and no issues.

miles267, 2019-10-04:
Thanks @semotech. I seem to have made progress with a different USB cable and the RP0. However I do noticed even with my 128 GB microSD card I’m seeing a sentry icon with a red exclamation mark symbol in only a couple of hours. Should I not be using the default .conf camsize=100% setting? Even if I am only using the SD card for video and not any music?

marcone, 2019-10-04:
With v10 and a 128GB card, I'd use a size of "24G" (literally specify that as the size, don't use percentages).
That's enough for 2 hours of Sentry recordings between archiving (i.e. 12 Sentry events). Add 8 for every additional hour you think you need.

miles267, 2019-10-04:
> With v10 and a 128GB card, I'd use a size of "24G" (literally specify that as the size, don't use percentages).
> That's enough for 2 hours of Sentry recordings between archiving (i.e. 12 Sentry events). Add 8 for every additional hour you think you need.

Thanks @marcone.  But if I'm only designating 24G (of 128G total), how will it utilize the remaining 104G of available space on the SD card?  Also if I used the default setting of camsize=100%, do I now have to completely reimage the card and run setup all over again with a new .conf to set it up to apply the new export camsize=24G?

marcone, 2019-10-04:
First off, you're overestimating how much free space there will actually be. 😁
A '128GB' card is generally around 128 billion bytes, or about 119 "real" gigabytes (the kind measured in powers of 2)
Then the OS takes about 2 gigabytes of that, and the rest is available for emulating a drive for the car.
In any case, what the free space is used for is taking snapshots of the disk image so that it can retain more than 1 hour of recordings. You can access these via Samba.
You can certainly use all the available space for the car, but it will end up being mostly unused.

Vipercat, 2019-10-05:
I get the warning 'low space for snapshots'. Is there a way to repartition or resize the CAM and MUSIC percentage to provide enough space for snapshots? To be honest, I don't understand where the size for /backingfiles/snapshots is based upon, because when I do a df -h (see below), then it seems like 38GB is allocated for snapshots but I would not know where these 38GB came from.
root@TeslaRaspbian:~/bin# df -h
Filesystem                         Size  Used Avail Use% Mounted on
/dev/root                          1.6G  1.2G  365M  77% /
devtmpfs                           213M     0  213M   0% /dev
tmpfs                              217M     0  217M   0% /dev/shm
tmpfs                              217M  5.9M  211M   3% /run
tmpfs                              5.0M  4.0K  5.0M   1% /run/lock
tmpfs                              217M     0  217M   0% /sys/fs/cgroup
tmpfs                              217M     0  217M   0% /var/tmp
tmpfs                              217M   16K  217M   1% /var/log
tmpfs                              217M   48K  217M   1% /tmp
/dev/mmcblk0p1                      44M   23M   21M  52% /boot
/dev/mmcblk0p4                      94M  1.6M   86M   2% /mutable
/dev/mmcblk0p3                      58G   52G  6.0G  90% /backingfiles
//diskstation/TeslaUSB/TeslaMusic   44T  7.6T   37T  18% /mnt/musicarchive
//diskstation/TeslaUSB/TeslaCam     44T  7.6T   37T  18% /mnt/archive
/dev/loop0p1                        38G  455M   38G   2% /backingfiles/snapshots/snap-000000/mnt
tmpfs                               44M     0   44M   0% /run/user/1000
/dev/loop1                          14G  532M   14G   4% /mnt/music

marcone, 2019-10-05:
Don't worry about the snapshots, those messages are for diagnostics, they don't necessarily indicate anything's wrong.
But if you want to tweak sizes to make more efficient use of space, then reflashing is easiest. I recommend using absolute values, not percentages, e.g. "16G". I see you're only using 532MB of music, so music could be set to "1G".

Vipercat, 2019-10-05:
@marcone, thanks again for your quick response. The reason the music folder is only using 532MB is because an update of the installation (not a reflash) resulted in the rebuild of the music.bin file, so I have to copy my music all over again from my NAS to the music folder. Would be nice if a refresh or update would not result in this. Furthermore, I prefer not to do a reflash because I've also installed a webserver on the RPI to view footage in the RecentClips from my PC when the RPI is connected, and the installation of the webserver was not without hassle, so I prefer not to do that all over again.