marcone/teslausb #270: Tests went okay but after delivery data transfer doesn't seem to work

Helmi, 2019-09-23:
Hey there,

I only got my Model 3 today so please bear with me if some of that sounds stupid. I took my prepared Pi with teslausb with me and did test the music syncing before which worked fine. Also Pushover pushes worked well.

When putting the thing in today the car instantly offered sentry mode and dashcam and I took a few recordings which are on the drive and look okay via SMB.

What didn't work is the sync with my server. I'm quite tech savvy and I'm certain that the Wifi connection works (actually pulled diagnostics via SSH while the thing is still plugged in the car) and the Server (Synology DS) also works fine as it did before.

wonder what i'm doing wrong (I'm sure it's me) and if someone could help.

Here's the diagnostics: https://termbin.com/rj47

Car is on 2019.32.2.2 da05838 in case that matters

Thanks for helping!

marcone, 2019-09-23:
Your diagnostics appear to be incomplete/truncated. They end on September 10 and don't show the sections after the archiveloop log.

Helmi, 2019-09-24:
sorry, termibin seems to have an issue with the size and I didn't recognize: https://clbin.com/URGwp

marcone, 2019-09-24:
One thing that looks a bit suspicious to me is that you're connected to a 10.10.10.* network:
```
wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.10.10.35  netmask 255.255.255.0  broadcast 10.10.10.255
```
but your archive server is on a different network:
```
//192.168.88.10/Tesla /mnt/archive cifs credentials=/root/.teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777,vers=2.0, 0
```
and that combined with the fact that the Pi is continuously trying to connect to the archive server and never getting a connection:
```
Mon 23 Sep 20:30:32 CEST 2019: Sleeping before retry...
Mon 23 Sep 20:30:33 CEST 2019: Retrying...
Mon 23 Sep 20:30:36 CEST 2019: Sleeping before retry...
Mon 23 Sep 20:30:37 CEST 2019: Retrying...
Mon 23 Sep 20:30:40 CEST 2019: Sleeping before retry...
Mon 23 Sep 20:30:41 CEST 2019: Retrying...
Mon 23 Sep 20:30:44 CEST 2019: Sleeping before retry...
Mon 23 Sep 20:30:45 CEST 2019: Retrying...
(etc)
```
points at a possible configuration problem.

Also, it's odd that it just keeps retrying for days on end like that. It's supposed to try only 10 times and then log "Attempts exhausted" and stop: https://github.com/marcone/teslausb/blob/main-dev/run/archiveloop#L117
Did you modify the code in any way?

Helmi, 2019-09-24:
no, no modifications from my side. The IP reachability shouldn't be an issue. The router routes those networks. The 10.10.50.0/24 only has servers, my machine is also on the 10.10.10.0/24 network and there is no issue getting there. But good to know the issue is on that end - I somehow couldn't identify that from the log. I see what I can find there.

Would still be interesting to know why it keeps trying endlessly... 🤔

marcone, 2019-09-24:
I'm still puzzled as to why it seems to try an infinite number of times instead of stopping after 10 tries like it should, but on closer look, it seems like it actually tried to connect to the archive, then failed:
```
Mon 23 Sep 19:36:26 CEST 2019: Waiting for archive to be reachable...
Mon 23 Sep 19:36:35 CEST 2019: Archive is reachable.
Mon 23 Sep 19:36:35 CEST 2019: Waiting for time to be set by ntpd...
Mon 23 Sep 19:36:36 CEST 2019: Time now set
```
then it takes a snapshot of the current disk image before proceeding, but then fails to actually mount the archive:
```
Mon 23 Sep 19:38:22 CEST 2019: Mounting /mnt/archive...
Unable to find suitable address.
Mon 23 Sep 19:38:27 CEST 2019: Failed to mount /mnt/archive.
Mon 23 Sep 19:38:27 CEST 2019: Attempts exhausted.
Mon 23 Sep 19:38:27 CEST 2019: Failed to mount cam archive.
Mon 23 Sep 19:38:27 CEST 2019: Couldn't connect archive, skipping archive step
```
then after this is where it starts to wait forever:
```
Mon 23 Sep 19:38:27 CEST 2019: Connecting usb to host...
Mon 23 Sep 19:38:27 CEST 2019: Connected usb to host.
Mon 23 Sep 19:38:28 CEST 2019: Truncating log...
Mon 23 Sep 19:38:28 CEST 2019: Waiting for archive to be unreachable...
Mon 23 Sep 19:38:29 CEST 2019: Sleeping before retry...
Mon 23 Sep 19:38:30 CEST 2019: Retrying...
Mon 23 Sep 19:38:32 CEST 2019: Sleeping before retry...
Mon 23 Sep 19:38:33 CEST 2019: Retrying...
Mon 23 Sep 19:38:37 CEST 2019: Sleeping before retry...
Mon 23 Sep 19:38:38 CEST 2019: Retrying...
Mon 23 Sep 19:38:41 CEST 2019: Sleeping before retry...
Mon 23 Sep 19:38:42 CEST 2019: Retrying...
Mon 23 Sep 19:38:45 CEST 2019: Sleeping before retry...
Mon 23 Sep 19:38:46 CEST 2019: Retrying...
Mon 23 Sep 19:38:49 CEST 2019: Sleeping before retry...
Mon 23 Sep 19:38:50 CEST 2019: Retrying...
```
So it looks like it's actually waiting for the archive to be unreachable.
Now the way it determines whether the archive is reachable is by pinging the archive server. So it looks like something is responding to those pings, making teslausb think the archive is reachable, but I suspect that whatever's responding to those pings isn't actually the archive server, since the mount failed with that "Unable to find suitable address" message.

marcone, 2019-09-24:
(you could ssh into the pi and try to "mount /mnt/archive" manually to see if that works)

Helmi, 2019-09-24:
```
root@teslausb-Snoopy:~# mount /mnt/archive
Unable to find suitable address.
´´´

and the ping:

```
root@teslausb-Snoopy:~# ping 10.10.50.50
PING 10.10.50.50 (10.10.50.50) 56(84) bytes of data.
64 bytes from 10.10.50.50: icmp_seq=1 ttl=63 time=4.94 ms
64 bytes from 10.10.50.50: icmp_seq=2 ttl=63 time=7.24 ms
64 bytes from 10.10.50.50: icmp_seq=3 ttl=63 time=8.08 m
´´´

so the server is definitely reachable. I tried mounting manually but couldn't get it to work. May this be an issue with the read only file system? I was also trying to run `/root/bin/setup-teslausb`  as diagnostics.txt suggested with not so much success:

```
root@teslausb-Snoopy:~# /root/bin/setup-teslausb
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
Read setup info from teslausb_setup_variables.conf (yes/no/cancel)? yes
reading config from /root/teslausb_setup_variables.conf
/root/bin/setup-teslausb is up to date
Updating package index files...
Ign:1 http://archive.raspberrypi.org/debian stretch InRelease
Ign:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Err:3 http://archive.raspberrypi.org/debian stretch Release
  Could not open file /var/lib/apt/lists/partial/archive.raspberrypi.org_debian_dists_stretch_Release - open (30: Read-only file system) [IP: 93.93.135.188 80]
Err:4 http://raspbian.raspberrypi.org/raspbian stretch Release
  Could not open file /var/lib/apt/lists/partial/raspbian.raspberrypi.org_raspbian_dists_stretch_Release - open (30: Read-only file system) [IP: 93.93.128.193 80]
Reading package lists... Done
W: chown to _apt:root of directory /var/lib/apt/lists/partial failed - SetupAPTPartialDirectory (30: Read-only file system)
W: chmod 0700 of directory /var/lib/apt/lists/partial failed - SetupAPTPartialDirectory (30: Read-only file system)
W: Not using locking for read only lock file /var/lib/apt/lists/lock
W: Problem unlinking the file /var/lib/apt/lists/partial/.apt-acquire-privs-test.PwkrJe - IsAccessibleBySandboxUser (30: Read-only file system)
W: Problem unlinking the file /var/lib/apt/lists/partial/.apt-acquire-privs-test.E1JZrd - IsAccessibleBySandboxUser (30: Read-only file system)
W: Problem unlinking the file /var/lib/apt/lists/partial/archive.raspberrypi.org_debian_dists_stretch_InRelease - PrepareFiles (30: Read-only file system)
W: Problem unlinking the file /var/lib/apt/lists/partial/raspbian.raspberrypi.org_raspbian_dists_stretch_InRelease - PrepareFiles (30: Read-only file system)
W: Problem unlinking the file /var/lib/apt/lists/partial/archive.raspberrypi.org_debian_dists_stretch_Release - PrepareFiles (30: Read-only file system)
E: The repository 'http://archive.raspberrypi.org/debian stretch Release' does no longer have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: Problem unlinking the file /var/lib/apt/lists/partial/raspbian.raspberrypi.org_raspbian_dists_stretch_Release - PrepareFiles (30: Read-only file system)
E: The repository 'http://raspbian.raspberrypi.org/raspbian stretch Release' does no longer have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: Problem unlinking the file /var/cache/apt/pkgcache.bin - RemoveCaches (30: Read-only file system)
W: Problem unlinking the file /var/cache/apt/srcpkgcache.bin - RemoveCaches (30: Read-only file system)
Failed, retrying
```

not sure if that's how it's meant to be or if there's really something going wrong with the setup. :-/

Helmi, 2019-09-24:
argh. better ignore me. Sorry for stealing your time. I just noticed the server IP was indeed wrong. I had changed it weeks ago but looks like the setup was done before. It indeed is on a different network but hence the mixup with 192.168.88.10 vs. 10.10.50.50 in the thread. 

I also found out how to set the FS to readwrite and am updating the system and configs now and hopefully will get it going. I'll get back if it still doesn' t work then.

Again sorry.

Helmi, 2019-09-26:
I need to reopen this unfortunately as I can't seem to get the archive server reconfigured.

What I did:

```
sudo -i
/root/bin/remountfs_rw
changed IP in teslausb_setup_variables.conf
ran /root/bin/setup-teslausb
```

Still doesn't work and the diagnostics file still has the old IP in the first section. I'm not sure though what the diagnostic file is meant to show as it also has log files until back a few days or weeks.

Is there anything I missed when reconfiguring? Any way to check which server is configured for archiving?

marcone, 2019-09-26:
You'll need to update /etc/fstab manually, the setup script won't update the existing one if you just changed the IP.

closed 2019-09-26 by marcone