marcone/teslausb #272: Drive not mounting properly at boot

Nicer82, 2019-09-26:
Used the prebuilt image to configure a Pi Zero W with an 128 GB SD card to use on my 2018 Tesla Model S.

First problem I hadis that the Pi is not getting enough voltage from a single USB port, which results into an endless reboot cycle. I fixed this by connecting both USB ports of the Pi to both USB ports in the front of the model S. 

The problem I have now is that the drive doesn't get mounted in the Tesla (DashCam icon doesn't show up) every time it needed to boot (when the car was not in Sentry mode and it went to sleep, cutting the power on the USB ports). The workaround I have is to unplug the data-usb port of the pi and insert it again after a few seconds, this way the dashcam icon shows up and everything works perfectly. The Pi doesn't reboot this way because it still gets power from the other USB port.

Anyone else experiencing this with model S/X? I am running 2019.32.2.1 in the Tesla. Also still have the old Tegra-based MCU in the car.

marcone, 2019-09-26:
https://github.com/marcone/teslausb/issues/35

closed 2019-09-26 by marcone

Nicer82, 2019-09-26:
@marcone I did, all I can see is: 'if you are using a separate power source (battery, 12V adapter, OBD2 adapter), ensure that the cable connected to the Pi's USB port is a data-only cable, and does not provide power.' but if I do that, the pi will go into the reboot cycle again. maybe an idea to clarify in the troubleshoot how to fix this one?

marcone, 2019-09-26:
https://github.com/marcone/teslausb/issues/35 also mentions providing diagnostics, which you didn't do.
In any case, from your description this seems more like a car problem, possibly because it's an older MCU. Have you tried putting a USB hub between the car and the Pi? That seems to help for some people.

Nicer82, 2019-09-26:
Will try the hub, if that doesnt work, I will submit diagnostics. Thx for the help.

Nicer82, 2019-09-27:
Just to confirm: the USB hub (unpowered) fixed the issue, I can now connect the Pi using one USB port without getting into the endless reboot loop and after boot the device is mounted properly now. Thx for the support!