marcone/teslausb #273: Slow wifi - how to increase timeout

tgerighty, 2019-09-27:
Where I park isn't getting me good wifi signal. I can get a connection, but the transfer is very slow. How can I increase the timeout since I currently get:

Fri 27 Sep 03:43:50 BST 2019: Creating output directory ''
Fri 27 Sep 03:43:50 BST 2019: Creating output directory '2019-09-22_19-22-03'
Fri 27 Sep 03:43:50 BST 2019: Moving '2019-09-22_19-22-03/2019-09-22_19-17-16-left_repeater.mp4'
Fri 27 Sep 03:44:11 BST 2019: connection dead, killing archive-clips
Fri 27 Sep 03:44:11 BST 2019: Unmounting /mnt/cam...
Fri 27 Sep 03:44:12 BST 2019: Failed to unmount /mnt/cam, trying lazy unmount.
Fri 27 Sep 03:44:12 BST 2019: lazily unmounted /mnt/cam
Fri 27 Sep 03:44:12 BST 2019: Finished archiving.

If I move the pi to the house, it connects and transfers fine so it's fully functional.

marcone, 2019-09-27:
When was the last time you ran setup? It currently decides the connection is dead if it has been unable to ping the server 10 times in a row (used to be 2 times).
Your connection has to be *really* bad if that's the case.
In any case, you can try modifying /root/bin/archive-clips.sh and change the "10" in the "connectionmonitor" function to a higher value.

tgerighty, 2019-09-27:
A couple of weeks ago. Will do an update and see if it works.

tgerighty, 2019-09-29:
yeah that worked. thx. this one can be closed.

closed 2019-09-29 by marcone