marcone/teslausb #276: Out of Space Errors

raushel, 2019-09-27:
If I leave my car at the airport for a few days with sentry mode enabled (100+ events), I notice that I can no longer run setup-teslausb and /dev/root has no free space. I'm going to need to reflash for v10 shortly, but wanted to see if there was anything else to prevent this going forward. Using camsize=100% if it makes a difference on a 128GB SD card.

[image]

./setup-teslausb as it throws no space left on device errors:

[image]

Diagnostics:
[diagnostics.txt](https://github.com/marcone/teslausb/files/3664749/diagnostics.txt)

marcone, 2019-09-28:
That's very strange, because the root partition is normally read-only, and so it can't fill up.
Did you remount it read-write to install some new software or something, and then forget to reboot afterwards?

raushel, 2019-09-28:
Unlikely, but certainly possible—always run halt after running updates to teslausb.

Had it happen on two different travel weeks, but re-flashed to fix between them.

marcone, 2019-10-25:
Nothing actionable here

closed 2019-10-25 by marcone