marcone/teslausb #277: WPA Passwords with slashes in them even when escaped make rc.local very sad

dopeghoti, 2019-09-28:
This may be rather complex to fix since you're using `sed` to write the `wpa_supplicant.conf` file, but the WPA password for my wireless network has `/`s in it, and that caused the first-boot to fail when running `rc.local`:

```
rc.local[464]: sed: -e expression #1, char 19: unknown option to `s'
[FAILED] Failed to start /etc/rc.local Compatibility
```

I was able to work around this by manually adding my password to the partially written WPA config file, booting, and manually `touch`ing `WIFI_ENABLED`; I was then able to SSH in and manually run `/etc/rc.local` which seemed to kick things back off successfully.

Things I tried in `teslausb_setup_variables.conf` given an example WPA password of `secret:pass/word` include but are not limited to:

```
export WIFIPASS='secret:pass/word'
export WIFIPASS='secret:pass\/word'
export WIFIPASS='secret\:pass\/word'
export WIFIPASS=secret:pass/word
export WIFIPASS=secret:pass\/word
export WIFIPASS=secret\:pass\/word
```

marcone, 2019-09-29:
Weird, I would expect an escaped / to work:
```
pi@teslausb2:~ $ export WIFIPASS='foo\/bar'
pi@teslausb2:~ $ echo TEMPPASS | sed -e "s/TEMPPASS/${WIFIPASS}/g"
foo/bar
```
The "unknown option to 's'" is what I get when *not* escaping the /:
```
pi@teslausb2:~ $ export WIFIPASS='foo/bar'
pi@teslausb2:~ $ echo TEMPPASS | sed -e "s/TEMPPASS/${WIFIPASS}/g"
sed: -e expression #1, char 16: unknown option to `s'
```

dopeghoti, 2019-09-29:
It _might_ be the escapes being lost when `source`ing the conf file, possibly?

This can be worked around by not using `/` for the delimiter for the `sed` command.  `sed` will use whichever character that follows `s` for the delimiter, including a character (such at a control character) that cannot be part of a WPA password or SSID:

```
ghoti~/scratch$ printf "TEMPPASS\n" | sed "s/TEMPPASS/${WIFIPASS}/"
sed: -e expression #1, char 16: unknown option to `s'
ghoti:~/scratch$ printf "TEMPPASS\n" | sed "s_TEMPPASS_${WIFIPASS}_"
foo/bar
ghoti:~/scratch$ printf "TEMPPASS\n" | sed "s^GTEMPPASS^G${WIFIPASS}^G"
foo/bar
```
In the third example, I am using a literal Control-G character.  This would be a pretty simple PR; I will submit it shortly.

PR #279 (2019-09-29): More robust Wireless Network config
Use a non-printable character (^G) as the sed command delimiter.  Should resolve #277 

(The nonprintables are not visible in the diff here, but they are present in the files)
closed 2019-09-29 by marcone