marcone/teslausb #278: TeslaUSB Hangs in V10

DonC52, 2019-09-29:
Hi Marcone...

Ran into an interesting thing. I re-created the TeslaUSB on my Pi again and it seemed to work great. However, when I got home, I noticed that the light on the Pi was on SOLID. Not a good sign.

V10 changed where they save files now:
SentryClips - Holds the Sentry clips and can be deleted if room is needed by the car.
SavedClips  - Now holds the clips where you press the Cam Icon on the cars display.
RecentClips - Same as it use to be.

I get a the hang when it tries to link the first SentryClips file. I am attaching the diagnostic file here as well. [diag.txt](https://github.com/marcone/teslausb/files/3666197/diag.txt)

You got any solution for this issue? Does the V10 branch handle this issue?

Thanks....  (Hope you got V10 - it fixes a lot of issues I have in AP and NavAP.)

mattster98, 2019-09-29:
Confirming not working.  Seems to fail in the linking step(s).. I haven't updated in a while so this part is new to me.

Running archiveloop manually yields this error:

/root/bin/make_snapshot.sh: line 60: collection: unbound variable

Doesn't seem to like $collection.. not sure what that does.  All the other similar lines use a different variable.

marcone, 2019-09-29:
hmm, looks like something went wrong there when I copied that change off my device. Should be fixed now..

closed 2019-09-29 by marcone