marcone/teslausb #282: Small bug updating SSID name = fail

SemoTech, 2019-09-30:
Heads up there is a small bug I found as follows:

Remounting the FS and editing the config file to change the SSID AP Name will not have any effect, as the SSID name stored in /etc/hostapd/hostapd.conf is not updated when running /root/bin/setup-teslausb selfupdate and /root/bin/setup-teslausb

Not a critical bug but I am sure easy to fix.

SemoTech, 2019-09-30:
Also noticed the channel is hard set in /etc/hostapd/hostapd.conf  
Mine was on channel=11 
I wonder if it's fixed or changes as needed based on interference?

SemoTech, 2019-09-30:
Maybe we can add this to  /etc/hostapd/hostapd.conf to improve the AP mode WiFi:

channel=0
acs_num_scans=5
acs_chan_bias=1:0.8 6:0.8 11:0.8
dtim_period=2
chanlist=1 6 11
beacon_int=100
max_num_sta=15

marcone, 2019-09-30:
The channel in hostapd.conf is normally ignored: since the radio can't use 2 channels at the same time, it will use the same channel that the Pi itself is connected to.
Not sure what it does when you're not on wifi, but I assume it'll just keep using that same channel.

Can you explain why these settings would improve anything? (for example the values for dtim_period and beacon_int you listed appear to be the default values, so it doesn't seem necessary to specify them, and max_num_sta is how many stations it will keep in its internal list, and doesn't seem like it would affect performance)

SemoTech, 2019-09-30:
I see. Good to know.  It just seemed like there was room for improvement with the 2.4GHz options.  Once I saw that changing the SSID name in the config file did not rename the AP name I was digging and found these options.  Thought they could help improve performance.

marcone, 2019-11-12:
Happy to consider pull requests, but I won't be working on this. I don't think it's worthwhile to cover all these edge cases (it's not just the two SSIDs that could change, but also the password and IP address. If people want to change any of these after the fact, they can either update the relevant configuration files, or reflash and start over.

closed 2019-11-12 by marcone