marcone/teslausb #286: teslausb disconnects after 5-30 minutes of recording

elight, 2019-09-30:
I’m driving, the dashcam appears to be working, and then suddenly the camera icon disappears.

I check the Pi and I’m getting a single blink of the LED every second or so, rhythmically.

I’ve tried swapping USB cables to no avail.

My device is on a splitter so I tried without the splitter. Sometimes, same failure. Sometimes I get the black X and an error message about USB write speed being inadequate.

I’m at wit’s end.

Is there a way to put teslausb into debug mode so that each script run dumps output to logs?

I haven’t yet set up WiFi to allow me to ssh in and poke around. I’m hoping to do that as soon as I can get to a shop that’ll sell me some cables I need.

Thoughts?

aberen001, 2019-10-01:
Which version of teslausb are you using, and which dev branch?

elight, 2019-10-01:
I’m on the v10 branch now.

marcone, 2019-10-01:
The v10 branch is outdated. Please use main-dev.
I'll be deleting the v10 branch soon to avoid further confusion.

aberen001, 2019-10-01:
Will reformat to main-dev. Which pre-built image is recommended for Tesla update 2019.32.11? Buster or teslausb-20190615 ?

marcone, 2019-10-01:
Buster or Stretch shouldn't really matter. Buster is still in the testing stage, but has been working OK for people it seems. Stretch can be considered stable, but you won't be able to update it to Buster later without reflashing.

closed 2019-10-01 by elight