marcone/teslausb #287: Feature Request: Ability to delay archiving when home WiFi becomes available

fatfurrycat, 2019-09-30:
Currently, archiving occurs as soon as my home WiFi network is detected. 
This occurs before I can even park the car, as my WiFi signal reaches quite some way from the house. and results in both camera and music partitions being unmounted. 

Ideally, I'd like to ensure that any dashcam / sentry footage of me returning home is fully intact until after I've parked the car and gone into the house. 

Also, is it necessary to dismount the music drive immediately when archiving begins? 
Sometimes, especially if I'm listening to an audiobook, I like to sit in the car and finishing listening to a particular chapter before I head into the house. Right now, I can't do that.

bladec59, 2019-09-30:
There is already a setting in teslausb_setup_variables.conf to set that delay; currently defaults to 20 seconds....but you can change it to whatever you want.

marcone, 2019-09-30:
It already waits 20 seconds by default, but you can make it wait longer if you want.
It should be possible to unmount camera and music drives independently, but that's more complexity that I don't really feel like adding.

closed 2019-09-30 by marcone

fatfurrycat, 2019-10-01:
Thanks guys. Don't know how I missed the delay the thing.
Understand about keeping the complexity down.