marcone/teslausb #288: USB Write Speed too slow error

ice2032, 2019-10-02:
I'm getting this error after about an hour or so.  If I unplug the Pi and plug back in, it will work for another short trip and then go back to the same error.  The SD card I'm using is a high quality, Class 10 Samsung endurance model so I'm 100% certain it's more than capable of the write speeds required.

miles267, 2019-10-02:
This degradation in write speeds on high perf SD cards seems to be a recurring theme. Planning to do a new build today on a Sandisk Extreme Pro ultra high speed card with the latest buster code, unless I should just hold off until this is addressed first.

marcone, 2019-10-02:
https://github.com/marcone/teslausb/issues/239