marcone/teslausb #289: Unable to mount mnt/archive

tomstacey, 2019-10-02:
Odd error on my TeslaUSB. The Pi can access the SMB server and is recording files to the CAM drive and copying to backing files but I get this error repeatedly when it tries to archive -

mount error: could not resolve address for raspberrypi: Unknown error
Wed  2 Oct 20:31:23 BST 2019: Failed to mount /mnt/archive.
Wed  2 Oct 20:31:23 BST 2019: Sleeping before retry...
Wed  2 Oct 20:31:24 BST 2019: Retrying...
Wed  2 Oct 20:31:24 BST 2019: Mounting /mnt/archive...
mount error: could not resolve address for raspberrypi: Unknown error
Wed  2 Oct 20:31:24 BST 2019: Failed to mount /mnt/archive.
Wed  2 Oct 20:31:24 BST 2019: Sleeping before retry...
Wed  2 Oct 20:31:25 BST 2019: Retrying...
Wed  2 Oct 20:31:25 BST 2019: Mounting /mnt/archive...
mount error: could not resolve address for raspberrypi: Unknown error
Wed  2 Oct 20:31:26 BST 2019: Failed to mount /mnt/archive.
Wed  2 Oct 20:31:26 BST 2019: Sleeping before retry...
Wed  2 Oct 20:31:27 BST 2019: Retrying...
Wed  2 Oct 20:31:27 BST 2019: Mounting /mnt/archive...
mount error: could not resolve address for raspberrypi: Unknown error
Wed  2 Oct 20:31:27 BST 2019: Failed to mount /mnt/archive.
Wed  2 Oct 20:31:27 BST 2019: Sleeping before retry...
Wed  2 Oct 20:31:28 BST 2019: Retrying...
Wed  2 Oct 20:31:28 BST 2019: Mounting /mnt/archive...
mount error: could not resolve address for raspberrypi: Unknown error
Wed  2 Oct 20:31:28 BST 2019: Failed to mount /mnt/archive.
Wed  2 Oct 20:31:28 BST 2019: Attempts exhausted.
Wed  2 Oct 20:31:28 BST 2019: Failed to mount cam archive.
Wed  2 Oct 20:31:28 BST 2019: Couldn't connect archive, skipping archive step
Wed  2 Oct 20:31:28 BST 2019: Connecting usb to host...
Wed  2 Oct 20:31:28 BST 2019: Connected usb to host.
Wed  2 Oct 20:31:28 BST 2019: Waiting for archive to be unreachable...

marcone, 2019-10-02:
What do you mean by "The Pi can access the SMB server"? The logs seem to be indicating otherwise. Is your SMB server called "raspberrypi" ?

tomstacey, 2019-10-03:
It is, although in the setup variables file, I have the SMB servers’s IP address,  not hostname.

marcone, 2019-10-03:
OK, not sure why it is referring to the name instead of the IP then. Diagnostics might show more, but this seems like a network/configuration problem to me.

tomstacey, 2019-10-03:
I’ll reinstall and try again!

Thanks for your help.

tomstacey, 2019-10-03:
Hi Marcone,

Can you close this - a reinstall fixed it

closed 2019-10-03 by marcone