marcone/teslausb #290: Discord Spam Ban Wave (also server missing)

raushel, 2019-10-03:
Well, my discord account (and a number of others got banned tonight via spam filter). Looks like the teslausb discord is no longer available (and invite broken).

If you're having problems, check this thread on reddit & email abuse@discordapp.com to get your account re-enabled, but also PM u/karrdian your ticket number.

https://old.reddit.com/r/discordapp/comments/dcemhh/discord_accounts_getting_disabledno_emails_from/

Not sure about getting the server re-instated, but it's definitely been removed from my server list.

raushel, 2019-10-03:
[image]

raz4725, 2019-10-03:
Wow, of all the servers in this ban wave, THIS ONE GOT HIT?

If you use the form (and possibly DM the guy in the thread) then they work pretty fast to unban the impacted folks. It took me ~2h to get unbanned. 

Note:
Everyone who was in the server has ~2 weeks before disabled accounts become permanently removed.

raushel, 2019-10-03:
I replied back to my support request after getting unbanned asking about getting the server re-instated, not sure, but @marcone may have to request it as the server owner after getting unbanned.

marcone, 2019-10-03:
Yes, I noticed my Discord account was apparently disabled and the server gone. I've contacted support but haven't heard back yet.

dopeghoti, 2019-10-03:
Wow, that really sucks. My account was also disabled, and I'm still waiting
on any response to my support ticket. Here's hoping everything gets
reversed.

raushel, 2019-10-03:
Send a PM to: https://old.reddit.com/user/karrdian with your ticket number / discord email address. You'll get taken care of quickly.

raz4725, 2019-10-03:
Check the Reddit thread. You should use the abuse/trust and safety form and get a ticket number, then DM the Discord guy on Reddit. Regular support may not respond to this.

-------- Original Message --------

dopeghoti, 2019-10-03:
Already have. No response yet, but thanks for looking out (:

gacevedo, 2019-10-03:
Same thing happened to me. I just emailed abuse@discord.com. Will see what happens.

gacevedo, 2019-10-03:
On that note. Why Discord? Generally that's for gaming. Maybe it's time to migrate to Slack? Or maybe IRC. Good times way back then.

marcone, 2019-10-03:
My account has been restored, but the server is gone. Waiting for support to tell me whether they can resurrect it too, or whether I need to create a new one.
As for why Discord, mainly because I have a Discord account already and am somewhat familiar with it. Slack would require me to sign up for Yet Another Account.

marcone, 2019-10-03:
Well the plot thickens. In response to my question about restoring the server, I got this response from Discord:

"We'd like to remind you that doing things that are questionably legal on Discord is not a good idea, including, for example, automating or botting goods in a way that a company might decide to file a lawsuit over"

That's all they said.
It might be time to move to Slack or some other forum instead after all, since Discord is clearly a bit... disorganized.

gacevedo, 2019-10-03:
Yeah, I haven't heard from them (Discord) regarding my account. I get what you mean by "yet another account". Unfortunately, Discord professionalism is definitely sub-par. It might be time to look for an alternative.

I like Slack because it provides a feature called "threads" (or conversations), so even though you only get 1 free channel (I think), you can keep it organized using threads. Another really nice feature is the many integrations it offers, such as https://slack.github.com/.

Jonnehs, 2019-10-03:
Yeah my account got suspended too, they told me "i got caught in the spam filter somehow" and the T server is gone.. seems we triggered some banhammer algorithm

SemoTech, 2019-10-03:
Same here, just tried logging-in, and account was blocked.  Sent an email to support and received a reply in 5 min with:

> Hello,
Thanks for writing in. It looks like you got caught in our spam filter. After further review of your situation, I've gone ahead and lifted the suspension on your Discord account at this time.
Sincerely,
Discord Trust & Safety

However as @marcone already pointed out the server at https://discord.gg/kqN7FWk is gone...
Guess we can all just check here to see if it gets restored or @marcone makes a new one...

DMBlakeley, 2019-10-03:
My account just got restored. Same message as @SemoTech.

raushel, 2019-10-03:
Here's the response I got on the server. Looks like you can re-create it:

> Thanks for reaching out. We've been looking into this issue. It looks like the server was flagged errantly by our system for having an extremely high ratio of spammy/malicious accounts. We've suspended this system and are reviewing data to prevent this from happening again. 

> Unfortunately, there isn't any way for us to restore the server to a previous state, but you are free to make a new one. I'm deeply sorry for the inconvenience.

marcone, 2019-10-03:
I created a new one: https://discord.gg/b4MHf2x

closed 2019-10-03 by marcone